google/OpenSK #2: Why Nordic?

tcurdt, 2020-01-31:
It would be great to outline in the README why these two boards were picked.
And whether it should also work on other boards (and which).

jmichelp, 2020-01-31:
We selected the Nordic chip because it natively supports secure cryptographic acceleration, supports the three transport protocols described in the FIDO2 specification (USB, NFC and BLE) and has affordable development boards that are readily available for people interested in contributing to the project. On top of that it was already supported by Tock OS,  which made the development faster.

As explained in #3 supporting a custom board design around the same chip is fairly easy to do. Supporting another chip than the Nordic nRF52840 is also possible but requires more work (e.g. implementing the USB interface on Tock OS for the chip).

tcurdt, 2020-01-31:
Thanks! I think it would be super helpful to have this in README or some FAQs.

eric-appion, 2020-01-31:
It appears to me that the actual ARM CryptCell 310 specifications are not publicly available, and it is therefore not possible to support it with open source software.

brouhaha, 2020-01-31:
It appears to me that the actual ARM CryptCell 310 specifications are not publicly available, and it is therefore not possible to support it with open source software.

solder-soldier, 2020-02-11:
> It appears to me that the actual ARM CryptCell 310 specifications are not publicly available, and it is therefore not possible to support it with open source software.

You can find Nordic's manual for it here: https://infocenter.nordicsemi.com/topic/ps_nrf52840/cryptocell.html?cp=4_0_0_5_5

Several examples are in the SDK: https://infocenter.nordicsemi.com/topic/sdk_nrf5_v16.0.0/crypto_examples.html?cp=7_1_4_3

brouhaha, 2020-02-11:
I'm well aware of that manual, which confirms exactly what I said: "The CRYPTOCELL state is controlled via a register interface. The cryptographic functions of CRYPTOCELL are accessible by using a software library provided in the device SDK, not directly via a register interface."

Neither Nordic nor ARM provide documentation for the registers, just the proprietary closed-source binary library. As I said, it appears to not be possible to support the CryptoCell 310 with open source software.

marcan, 2020-03-12:
Looks like the CRYPTOCELL 312 SDK is open source. Maybe it can be adapted for the 310?

https://github.com/ARM-software/cryptocell-312-runtime

jmichelp, 2020-03-13:
That's what I've been using since August to talk to the CryptoCell.

There are a few differences between CC310 and CC312 though which need to be addressed. My current understanding is that the CC310 is a striped down version of CC312 (e.g. no AES192 and no AES256) that is flavored for IoT (i.e. it contains additional registers compared to the CC312).

kaczmarczyck, 2023-11-09:
It is now possible to use OpenSK as a library. While the API is still not fixed, people are actively working on new `Env` implementations.
There is a separate issue for the Cryptocell, so I'm closing this issue.

closed 2023-11-09 by kaczmarczyck