google/OpenSK #3: NRF52840 boards?

tmikaeld, 2020-01-31:
Does the key need to be the Nordic NRF52840 dev dongle or does the key only need to have the NRF52840 onboard?

jmichelp, 2020-01-31:
The main requirement today is the Nordic nRF52840 SOC.

The two development boards that Nordic made around their chip (the NRF52840-DK and the NRF52840-Dongle) are already supported but it's pretty easy to adapt the code to support a custom design around the Nordic chip. You need 1 button and at least 1 LED to be wired to the chip.

Supporting another chip than the Nordic is also possible but requires more effort.

tmikaeld, 2020-01-31:
Seems great, unfortunately I know nothing of programming hardware.

But if someone else is interested, this dongle would be a perfect match:

https://github.com/makerdiary/nrf52840-mdk-usb-dongle

It has a really small form-factor and is available with a form-fit shell.

brouhaha, 2020-01-31:
Just FYI, the official Nordic nRF52840 dongle sells for $10, which is less than the price of the Makerdiary dongle. The design of the Nordic dongle is open. It doesn't come with a case, but if you have a 3D printer...

tmikaeld, 2020-01-31:
@brouhaha I looked at having the shell printed, but the shell is much larger, less durable and doesn't have a ring to have it in a key-chain. Plus, the MD dongle size is smaller, which I prefer - I'm too used to the smaller form-factor of Yubikeys.

xiongyihui, 2020-02-01:
I managed to get it work with https://github.com/makerdiary/nrf52840-mdk-usb-dongle.
We just need to change the IOs. As I don't have a jlink, I used a CMSIS-DAP debug adaptor with `pyocd` to flash the firmwares.
The steps are a little different.

By the way, I use [gitpod.io](https://gitpod.io/#https://github.com/google/OpenSK) to build the project.

1.  Run `setup.sh`

2.  Change the pins of LEDs and Buttons at `OpenSK/third_party/tock/boards/nordic/nrf52840_dongle/src/main.rs`

    ```
    // The nRF52840 MDK USB Dongle LEDs
    const LED1_PIN: Pin = Pin::P0_04;           // not connected to a LED
    const LED2_R_PIN: Pin = Pin::P0_23;
    const LED2_G_PIN: Pin = Pin::P0_22;
    const LED2_B_PIN: Pin = Pin::P0_24;

    // The nRF52840 MDK USB Dongle button
    const BUTTON_PIN: Pin = Pin::P0_18;
    const BUTTON_RST_PIN: Pin = Pin::P0_03;     // not connected

    const UART_RTS: Pin = Pin::P0_05;
    const UART_TXD: Pin = Pin::P0_06;
    const UART_CTS: Pin = Pin::P0_07;
    const UART_RXD: Pin = Pin::P0_08;
    ```

3.  Run `board=nrf52840_dongle ./deploy.sh os` to build Tock OS

    The Tock OS firmware `nrf52840_dongle.elf` is at `third_party/tock/boards/nordic/nrf52840_dongle/target/thumbv7em-none-eabi/release/`
    
    ```
    gitpod /workspace/OpenSK $ ls third_party/tock/boards/nordic/nrf52840_dongle/target/thumbv7em-none-eabi/release/
    build  deps  examples  incremental  nrf52840_dongle  nrf52840_dongle.bin  nrf52840_dongle.d  nrf52840_dongle.elf
    ```
    
4.  Run `pyocd flash -t nrf52840 nrf52840_dongle.elf` to flash the Tock OS into the dongle.
5.  Run `board=nrf52840_dongle ./deploy.sh app` to build the OpenSK app

    The app contains two parts `padding.bin` and `cortex-m4.tbf` at `target/tab/`
    
    ```
    gitpod /workspace/OpenSK $ ls target/tab/
    cortex-m4.elf  cortex-m4.tbf  ctap2.tab  padding.bin
    ```

6.  Run `pyocd flash -t nrf52840 --format bin -a 0x40000 cortex-m4.tbf` and `pyocd flash -t nrf52840 --format bin -a 0x30000 padding.bin` to flash the app

     `cortex-m4.tbf` should be written to the address `0x40000`. `padding.bin` should be written to the address `0x30000`.

mikejrh, 2020-02-01:
Is it possible to program the NRF52840 USB Dongle as a security key without additional hardware ?

rmm200, 2020-02-01:
Google is presenting OpenSK as a tool for developers and hobbyists.
The Nordic Dongle costs about $10.
The associated hardware to program it costs roughly $1000, per installation guide.
Try to come up with a cheaper solution for us hobbyists. $1000 is more than I spent on my 3D printer.

brouhaha, 2020-02-01:
There are a multitude of options for programming the nRF52, none of which cost $1000. The Nordic nRF52 DK board can be used, and sells for $40.

rmm200, 2020-02-01:
It looks like nRF Connect Programmer can do the job:
https://devzone.nordicsemi.com/nordic/short-range-guides/b/getting-started/posts/nrf52840-dongle-programming-tutorial
Maybe the install guide could reference this?  All it shows are the very expensive JProbe accessories.

elfmimi, 2020-02-01:
I tried RAYTAC MDBT50Q-RX and it worked.
https://www.raytac.com/product/ins.php?index_id=89

Note: you need to tweak pin settings for the led and the button. It would be safer to also change the pin for uart so that it won’t interfere with the reset pin.

rmm200, 2020-02-01:
It looks like the MDBT50Q-RX does not include a bootloader, so you also need a MDBT50Q-RX to do the programming? Sure beats anything JLink for cost though. The NRF52840 does appear to include the DFU Bootloader, so programming by USB is possible.

xiongyihui, 2020-02-02:
If you have two  nRF52840 MDK USB Dongle, you can use one as a CMSIS-DAP debug adapter with OpenOCD or pyOCD

See https://github.com/xiongyihui/mbed-CMSIS-DAP

rmm200, 2020-02-02:
Well that leaves open the question. Can you use the nRF52840 dongle bootloader to install OpenSK on itself, using just the one dongle - or does that destroy the bootloader? May be time to buy a second dongle...

elfmimi, 2020-02-02:
I think it can be done. I mean, it should be possible to install OpenSK with ordinary bootloader and without using any jtag/swd adapters.

xiongyihui, 2020-02-02:
We can modify `OpenSK/third_party/tock/boards/nordic/nrf52840_dongle/layout.ld` to make Tock OS compatible with the nordic open bootloader.

```
MEMORY
{
  rom (rx)  : ORIGIN = 0x00001000, LENGTH = 124K
  prog (rx) : ORIGIN = 0x00030000, LENGTH = 832K
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 256K
}

MPU_MIN_ALIGN = 8K;

INCLUDE ../../kernel_layout.ld
```

rmm200, 2020-02-02:
Man - you are getting there! Glad I asked!
Now to just not lose track of this information...
And - Thanks!

MatchstickUK, 2020-02-03:
> Google is presenting OpenSK as a tool for developers and hobbyists.
> The Nordic Dongle costs about $10.
> The associated hardware to program it costs roughly $1000, per installation guide.
> Try to come up with a cheaper solution for us hobbyists. $1000 is more than I spent on my 3D printer.

The larger nRF52840 DK board spec lists it as "comes with an on-board SEGGER J-Link debugger allowing programming and debugging both the on-board SoC and external SoCs through the debug out header" so could one of those be used to provide the required support to the dongle ?

jmichelp, 2020-02-03:
Thanks all for your contributions.

Regarding JLink probes, multiple things:

1. If you're not using them for commercial use, check if you're not eligible for their EDU (or EDU mini) version which is more affordable.

2. Of course you can use other probes to program the dongle. JLink is the one that was already supported in Tock OS (and more specifically the tockloader tool) and it's embedded in the Nordic nRF52840-DK development kit. We're mentioning this as a tested/validated solution.

Regarding using another probe than the JLink, I tried a while ago with a Black Magic Probe and OpenOCD without success. I haven't tried with pyOCD instead. I'll give it a try and add documentation about it.

Regarding supporting the Nordic DFU bootloader, that's something we will indeed think about. Because it will also spare the cost of the Tag-Connect cable (even though one can also use a breadboard and press fit headers on the castellated holes for SWD).

Finally about supporting the nRF52840 MDK board, it's for sure possible but as mentioned in the review of the pull request, we would prefer the boards to be added directly to Tock OS instead of having to maintain patches for this. The patches we currently have in our repository are changes that are in the process of being upstreamed and we hope that we will soon be able to fully remove them from our repository.

Raboo, 2020-02-03:
> We can modify `OpenSK/third_party/tock/boards/nordic/nrf52840_dongle/layout.ld` to make Tock OS compatible with the nordic open bootloader.

@xiongyihui would this mean you can program the dongle with nRF Connect Programmer?
It would be super awesome if you could make a PR for Tock OS to also support nordic open bootloader.

rmm200, 2020-02-03:
Let me second the Super Awesome. Us low budget users will thank you forever!

xiongyihui, 2020-02-04:
On nRF52840 MDK USB Dongle, we can install OpenSK without a debug adapter. See  [Get Started with OpenSK guide](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/getting-started/)

I still can not get OpenSK with Nordic open bootloader directly. However, the open bootloader supports self-upgrade, so we can replace open bootloader with a UF2 bootloader, and then get OpenSK work with the UF2 bootloader.

tmikaeld, 2020-02-04:
@xiongyihui Amazing! I'll start ordering a bunch :-D

gregvinyard, 2020-02-06:
Regarding the nRF Connect Programmer, it looks like you need hex files. The build script creates elf files.  Is there a way to convert elf to hex? Or modify the build script to output hex files?

MatchstickUK, 2020-02-06:
@gregvinyard the Building OpenSK page https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/building/ covers using bin2hex.py and hexmerge.py to built a single .hex file to install via UF2 bootloader.

I'm guessing someone has already tried using the output of that with nRF Connect Programmer ?

gregvinyard, 2020-02-06:
Thanks @MatchstickUK .  Apparently I have a lot to learn, heh.  I was able to convert the bin and tbf files to hex and flash them using nRF Connect Programmer, but OpenSK doesn't run.  I didn't brick the dongle, I can flash again.  No idea what I'm doing though. Time to do some more reading. :-)

rmm200, 2020-02-06:
Keep posting here please. You are ahead of me - and I am learning a lot.
I have found The OpenSK tools really do not like Windows - but WSL works really well.
Command line only (I loaded up Ubuntu 1804) and it is shaping up.
nRF Connect Programmer uses a lot of X libraries - does it require a GUI? That might get bounced over to Windows proper.

kaeltis, 2020-02-10:
Easier flashing of the nRF52840 Dongle would indeed be awesome.
I bought a JLink EDU Mini but can't figure out how to connect it to the dongle. I can't really find anyting that doesnt require soldering or a cable that is almost more expensive than the dongle and edu mini combined.

dennisg, 2020-02-14:
So... I got DFU flashing working for the standard nrf52840 dongle.

The comment @MatchstickUK made, the makerdiary wiki and most of the changes in PR #6 of course were instrumental for getting it to work!

A fork that works for me can be found at [dennisg/openSK](https://www.github.com/dennisg/openSK/tree/nrf52840_dongle_bootloader).

In the root I have added a script that leads you through the complete process: `flash_nrf52840_using_dfu.sh`.

notes: 
- I installed bin2hex.py and hexmerge.py in /usr/local/bin
- I use nrfutil to flash the hex, so using the original bootloader/ DFU that the dongle is shipped with

```
$ dmesg
...
[853099.670504] usb 1-2: New USB device found, idVendor=1915, idProduct=521f
[853099.670510] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[853099.670514] usb 1-2: Product: OpenSK
[853099.670519] usb 1-2: Manufacturer: Nordic Semiconductor ASA
[853099.670565] usb 1-2: SerialNumber: v0.1
[853099.673438] hid-generic 0003:1915:521F.001F: hiddev2,hidraw4: USB HID v1.10 Device [Nordic Semiconductor ASA OpenSK] on usb-0000:00:14.0-2/input0
```

dennisg, 2020-02-14:
mmhh... It seems that when I flash the dongl it works, but as soon as I unplug/ replug the dongle the bootloader enters DFU state immediately. Will need to do some research on that, still...

gregvinyard, 2020-02-14:
@dennisg That's so cool!  Making progress. :-)

Something I was planning on trying this weekend is using nrfutil to generate a DFU settings file.  From what I've read, DFU will look for an application to launch, and if it doesn't find one will put the device into DFU mode.  The DFU settings file tells DFU what memory address the application to launch is at.

So what I'm hoping, is if we set TockOS to live at 0x1000, and then padding at 0x30000 and OpenSK at 0x40000, and then create a DFU settings file that says the application to launch is at 0x1000, then we'll be good.  I don't really know how TockOS knows where to look for OpenSK though.  I'm assuming it will find it the same way it does when you flash the dongle without keeping DFU.  I don't know though, because I also read that TockOS expects certain things to exist starting at 0x00.  But, @xiongyihui was able to get TockOS working at 0x1000 with a UF2 bootloader, so I think it will be fine.

If you try this before me, let us know if it works!

rmm200, 2020-02-14:
@dennisg - I am so sorry - my lack of git-foo is showing. When I clone your repository, it does not have your recent updates, like flash_nrf52840_using_dfu.sh. How do I get your current updates? I know I could open that one - and copy - paste it - but that seems gauche.

elfmimi, 2020-02-14:
@rmm200 cd openSK ;: and then
git checkout nrf52840_dongle_bootloader

rmm200, 2020-02-14:
Whoot! You are a scholar and a gentleman! Worked like a charm!
Now I also have the main Google OpenSK cloned. Should I abandon that and
just use your branch?

dennisg, 2020-02-14:
Hey Robert,

I plan on creating a PR that hopefully pulls in all the required changes
into the main branch, eventually. Please do not abandon the main project.
Literally ALL the work done there
made this even remotely possible, I just joined together the intel hex-es
to flash it using the DFU, and it isn't working as planned yet...

D.

irtimmer, 2020-02-14:
When you flash using DFU a init packet is sent containing the information which is used for the bootloader settings. It contains a CRC value and app size which is used to validate the flashed data and is used to check the application during boot. I think the application state is part of the flashed image. Meaning after key generation the CRC value of the image is changed and OpenSK will no longer boot.

I'm not sure if my theory is completely correct but at least I found a workaround. As the flash is not wiped when flashing a new image you just need to flash only Tock after running flash_nrf52840_using_dfu.sh. This will change the app size and the CRC value in the bootloader, so only the Tock code is verified by the bootloader.

```
nrfutil pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application $OUTPUT/nrf52840_dongle.hex $OUTPUT/tock.zip
nrfutil dfu usb-serial -pkg $OUTPUT/nrf52840.zip -p /dev/ttyACM0
```

dennisg, 2020-02-15:
thanks @irtimmer, I've applied your suggestion to the script and can confirm it works

rmm200, 2020-02-15:
@dennisg As you make updates to your branch, git checkout nrf52840_dongle_bootloader will pick up the new changes?
And man - thanks to everyone participating in this effort. You will be helping a lot of people!

dennisg, 2020-02-15:
@rmm200, since you checked out the project already, use `git pull` while you are in the project folder

rmm200, 2020-02-15:
@dennisg Done. Hopefully that was the last of my stupid git questions. Now
I can stay in synch. And thanks.

zappos23, 2020-02-17:
Hi @dennisg, I tried cloning your repo and when I run the ./setup.sh to build the tock os, i got errors when it tries to apply the patch from your branch. if I skip the patch, I think I won't be getting the fixes you guys have made.. may I know what is the right steps to build the os and the app? Thanks

dennisg, 2020-02-17:
@zappos23 yes, my bad... sorry. I messed up the patch, but I hope that my last commit will fix it for you , so please `git pull` again and `./reset.sh` and `./setup.sh`.

zappos23, 2020-02-17:
@dennisg thanks a lot for the fix.. i will try it out

rmm200, 2020-02-17:
Worked for me. Thanks!

DG1001, 2020-02-18:
@dennisg tanks for your patches, works for me too!

zappos23, 2020-02-19:
Thank you all for the workaround. Got it working with the default bootloader.

mikejrh, 2020-02-23:
Is this now ready to install without additional hardware ? Could a newbie guide be provided ?  Thanks.

rmm200, 2020-02-23:
+1 on the request for a newbie guide...
I got the libraries and patches all installed. A few words on the next steps would sure help.

sissaias, 2020-02-24:
I bought the Nordic USB dongle to create a a FIDO key using OPENSK. I can't afford to spent £1000 to buy extra hardware. Is there another guide to program the board using just software tools like the Nordic Desktop Tools. Thanks

dennisg, 2020-02-26:
@sissaias  yes I did: have a look at [dennisg/OpenSK](https://github.com/dennisg/OpenSK/tree/nrf52840_dongle_bootloader). After the generic installation instructions, run the `flash_nrf52840_using_dfu.sh` script and if all's well your dongle will have been programmed correctly

sissaias, 2020-02-26:
> @sissaias yes I did: have a look at [dennisg/OpenSK](https://github.com/dennisg/OpenSK/tree/nrf52840_dongle_bootloader). After the generic installation instructions, run the `flash_nrf52840_using_dfu.sh` script and if all's well your dongle will have been programmed correctly

Thank you will give it a try tonight.

dennisg, 2020-02-29:
Hi @sissaias, I'm running this on linux, untested on OSX, sorry.

your root cause is this:
```
make: objcopy: No such file or directory
make: *** [target/thumbv7em-none-eabi/release/nrf52840_dongle.hex] Error 1
cp: third_party/tock/boards/nordic/nrf52840_dongle/target/thumbv7em-none-eabi/release/nrf52840_dongle.hex: No such file or directory
./deploy.sh: line 59: declare: -A: invalid option
```

which means that it cannot find `objcopy` used to generate the hex file.

A little bit further in the logs I see that when `deploy.sh` is run from the script, there seems to be a mismatch w.r.t. arguments fed into `declare` and lastly libusb-1.0 cannot be found.

I'm afraid I cannot help you to solve this since I do not have a mac for testing.

zappos23, 2020-02-29:
I had this problem on my MacOS too. Most probably it's due to old bash version not supporting declare option. 
I think you could either removed the declare in the sh file or update your bash. Thanks

jmichelp, 2020-02-29:
Portability issues should have been fixed. The `deploy.sh` script is gone.
If you're not using code from this repository I suggest you open an issue on the forked repository to tell them to update their fork

dennisg, 2020-03-02:
Hi @jmichelp, I updated the fork, still untested but I will hopefully soon find time to do so; are you interested in knowing how to flash a regular nrf52840 dongle with your application without the use of expensive hardware? Do you care to help with your knowledge to fix the 'double flashing' ? I would really appreciate your input!

sissaias, 2020-03-02:
Hello everyone

I upgraded bash as suggested now using @dennisg  fork I am getting the follow request for link:

[image]

mlberlin, 2020-03-03:
@dennisg awesome work! Many thanks! One small correction: the 1st argument of `deploy.py` in your bash script `flash_nrf52840_using_dfu.sh` should read `app` not `os`:
```
./deploy.py app --board=nrf52840_dongle
```
This probably also does away with your problem @sissaias.

jmichelp, 2020-03-03:
@dennisg Of course we're interested :) I actually bought myself 2 makerdiary dongles so that I can experiment/test.
I'll do my best to help merging your contribution to the main repository.

dennisg, 2020-03-04:
Yes, I've now started _really_ looking at the changes in master and indeed
we should build the `app` instead of the `os`. Thanks Adriaan.

Here's what I think we should fix to get flashing over DFU working again
(it's probably not that much):

- Within the `deploy.py` in `install_elf_file` the padding is installed at
address `0x30000` using tockloader just before flashing the app itself; we
should be able to generate a hex file that
contains that padding and that hex file should be merged with the os and
app hex-es.
- it might be best to create a completely separate target (e.g.
nrf52840_dongle_dfu) that incorporates these changes because we deviate
   quite a lot from the regular target
- fix the issue that we have to flash the os twice, maybe we just flash the
padding, os and app hexes separately? downside is that we then have to
reset the dongle 3 times during the process.

I'll try setting this up as soon as possible

dennisg, 2020-03-05:
@jmichelp Okay, I moved as much as possible/ seemed sensible from the bash script to the `deploy.py` and made it possible, using a `--dfu` flag, to bypass tockloader completely and only generate the hex files. Also updated the Tock OS main.rs for the nordic dongle (in most certainly a very dirty way, I'm no Rust developer) to set the table_offset_vector during compile time. Please have a look, sigh, do a face-palm and then point me to a better way to do it. thanks!

dennisg, 2020-03-05:
This is an overview of the memory layout if you load the hex files in nrf programmer.

[image]

[11 more replies]

PR #78 (2020-03-12): Revamp deploy.py
Now the script supports more flashing methods:
- JLink (with tockloader)
- OpenOCD (with tockloader)
- pyOCD
- Nordic DFU
- none (will produce an IntelHex file)

Also merged the contributions from:
- Yihui Xiong to support the Makerdiary USB dongle board
- Dennis Geurts to support Nordic DFU

Doc updated accordingly.

Imported 2 patches for Tock kernel:
- 06-add-set_vector_table_offset.patch (upstream tock/tock#1579)
- 07-nrf52-bootloader.patch (upstream tock/tock#1681)

Fixes #3 
Fixes #70 

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-03-12 by jmichelp