google/OpenSK #7: ARM CryptoCell-310

mcarrickscott, 2020-02-02:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version:
- Platform:

mcarrickscott, 2020-02-02:
The ARM CryptoCell-310 may be a bit of a problem. See this thread

https://devzone.nordicsemi.com/f/nordic-q-a/18578/arm-cryptocell-310-performance

which suggests that for elliptic curve cryptography at least , the hardware is actually slower than a software implementation. Also this thread

https://devzone.nordicsemi.com/f/nordic-q-a/46418/cryptocell-310-status-replacement 

suggests that it is already obsolete

Mike

nuno0529, 2020-02-11:
Another information for FIPS 140-2 certification on this nrf52840...
https://devzone.nordicsemi.com/f/nordic-q-a/39030/is-nrf52840-arm-cryptocell-310-fips-140-2-certified

OwensDataCenter, 2022-01-19:
Any updates?
I really want to see the ARM CryptoCell-310 be used.

kaczmarczyck, 2022-01-19:
Hi, thanks for your interest! I started using the Cryptocell for our custom bootloader, it's not submitted yet due to a shift in priorities. See [this commit](https://github.com/kaczmarczyck/OpenSK/commit/395b53285f23a01d5d4ad5cba467a31684d045a3) for example code to run SHA256. We don't have a PR lined up for the main OpenSK app yet though.

coelner, 2022-05-29:
I can't find anything about this here in this git: https://limitedresults.com/2020/06/nrf52-debug-resurrection-approtect-bypass/

https://infocenter.nordicsemi.com/pdf/in_133_v1.0.pdf

Not directly an issue, but maybe the cryptocell gains some attention back.