google/OpenSK #13: error[E0432]: unresolved import `alloc::alloc::AllocRef`

garyexplains, 2020-02-05:
When I run: board=nrf52840dk ./deploy.sh app os

I get the following:

```
No apps are installed on the board

   Compiling linked_list_allocator v0.6.6
error[E0432]: unresolved import `alloc::alloc::AllocRef`
  --> /home/gary/.cargo/registry/src/github.com-1ecc6299db9ec823/linked_list_allocator-0.6.6/src/lib.rs:14:20
   |
14 | use alloc::alloc::{AllocRef, AllocErr, Layout};
   |                    ^^^^^^^^ no `AllocRef` in `alloc`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `linked_list_allocator`.
```

joostd, 2020-02-05:
This should be fixed with https://github.com/google/OpenSK/pull/10 that was merged a couple of hours ago. Just update (or run setup.sh) to obtain the fix.

garyexplains, 2020-02-05:
Thanks, working now.

closed 2020-02-05 by garyexplains