google/OpenSK #15: NFC support?

Raboo, 2020-02-05:
I know it's early in the project, but is NFC support planned?
And how does that feature look priority wise?

jmichelp, 2020-02-05:
NFC support cannot happen on the Nordic dongle for 2 main reasons:

* there is no antenna and therefore this would require additional electronics to be soldered on it (antenna and 2 tuning capacitors)
* the Nordic nRF52840 chip requires more power than what the NFC field will provide. This means that we would need a battery to power the chip and the NFC field would wake-up the CPU in order to not drain the battery too quickly. This is technically possible but again requires extra circuitry.

This is technically possible to do it and design it, for example, on the nRF52840-DK development kit which has all the requirements (antenna and battery). But I'm not aware about an off-the-shelf board that would provide this in a portable form factor.

danrossi, 2020-02-24:
I have the same question. These dongles need just usb and NFC for FIDO authentication. Other wireless authentication is a failure, considering years of evidence both are exploitable .. I have no idea why it would be using wireless to connect to a desktop ? It just needs NFC tap for phones. 

Those boards are not so useful, and confused of it's purpose. Is there another one comparable to my current Yubikey to try ?

Raboo, 2020-02-29:
So support for NFC is not planned then? Or should this ticket be open as a reminder that some want NFC support? How has Yubico solved it in such a portable way?

farribeiro, 2020-03-01:
Isn't a similar usb token with cert (Brazilian or Estonia gov use that individual certs) when insert the SK, a password (or PIN) then login occur.

For the NFC I think as Google Authenticatior feature sounds but not for a Secure Key

danrossi, 2020-03-01:
I'm a Yubikey user. Have been for years. I use it for my windows login also. I have not tried the NFC tap on the phone but probably should try whatever supports it. 

Phone apps don't even support it. Facebook certainly doesn't ask for it when I need it on the desktop, neither does Gmail. So the phone is the backdoor into my stuff because of no Yubikey. 

Google Authenticator is a failure, there is rootkits to get access to that stuff now and resets with the phone. 

What I don't get about that dongle is why it has wireless capabilities known to have security issues. It shouldn't have wireless on here at all. It should have NFC. Is there a better dongle board to try ?

jmichelp, 2020-03-01:
It's not because the Nordic chip supports wireless protocols that they are enabled. At the moment they're not. Should we want to support them in the future, it's convenient to be able to do it with just a firmware update rather than having to completely change the hardware. And that was part of the choice for the Nordic chip. And NFC is a wireless protocol by the way :)

BigPictures, 2020-05-11:
[SoloKeys](https://github.com/solokeys/solo)' hardware would work for this. They are also working on a [new board](https://solokeys.com/blogs/news/update-on-our-new-and-upcoming-security-keys) based on the LPC55S69 that might work better with OpenSK.

danrossi, 2020-05-11:
@BigPictures solokeys looks amazing and the ticket. Both Bluetooth and Wifi have constant hardware security issues and a contradiction for a security key. So Solokeys has it's own firmware then and not suitable with OpenSK ? A hardware key should be usb and nfc only if anything.

jmichelp, 2020-05-11:
The LPC55D69 is still a Cortex-M chip so it's definitely possible to run Tock on it and then OpenSK. But because the chip is currently not supported out of the box, it's a substantial amount of work to add it: one has to write all the drivers in Tock in ordre to have buttons, LEDs, GPIOs, a console, a way to flash it, and in the case of OpenSK, adding USB.
NFC will require extra work both in the Tock OS kernel but also in the OpenSK application too in order to process the packets.

BigPictures, 2020-05-11:
I noticed after I wrote that that Tock OS has experimental support for an STM32 board -- SoloKeys currently uses an STM32L432. Anyway, it'd still likely be quite a bit of work.

jmichelp, 2020-08-05:
Good news on that front. NFC support will be added within the next months for the Nordic chip.

Caveats: although the Nordic chip internally supports NFC, it has the following 2 limitations:
1. it requires an additional antenna, which AFAICT is not provided by any of the commercially available USB dongle. But it's provided with the development kit.
1. the Nordic chip can't be powered through the NFC field and will require an additional battery. Again, not a problem for the development kit which comes with a coin cell battery but there's as of today no commercially available USB dongle form factored device which supports this.

maxfyk, 2020-08-09:
I recently found a NXP QN9080 USB dongle that has a built-in NFC antenna and has an Arm Cortex-M4F MCU. Theoretically, you can install OpenSK, add a small battery and get a compact NFC security key, but I have a feeling that I missed something.)
[image]

jmichelp, 2020-08-09:
By the shape and the look of it, the antenna is for Bluetooth, not for NFC.
NFC antennas must look like a coil (i.e. in the shape of a loop)