google/OpenSK #17: Can't install TockOS and the OpenSK application.

NaotoFujihiro, 2020-02-06:
## Expected Behavior
Complete the Installation [step2](https://github.com/NaotoFujihiro/OpenSK#installation).

## Actual Behavior
I got the following error.
> error[E0583]: file not found for module `key_material`
[board_nrf52840_dongle.log](https://github.com/google/OpenSK/files/4163504/board_nrf52840_dongle.log)

## Steps to Reproduce the Problem

1. `$ git clone --recursive git@github.com:google/OpenSK.git`
1. `$ ./setup.sh`
1. `$ board=nrf52840_dongle ./deploy.sh app os`

When I ran `./setup.sh`, `src/ctap/key_material.rs` wasn't generated.
[setup_sh.log](https://github.com/google/OpenSK/files/4163497/setup_sh.log)

## Specifications
I'm using pipenv.
```
$ pipenv --version
pipenv, version 2018.11.26
```

```
$ python --version
Python 3.7.6
```

```
$ rustc --version
rustc 1.42.0-nightly (f43c34a13 2020-02-02)
```

```
$ openssl version
LibreSSL 2.6.5
```

```
$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G2022
```

joostd, 2020-02-06:
Have you checked the output of setup.sh for error messages? On MacOS you need specific versions of gnu tools. See https://github.com/google/OpenSK/issues/11

jmichelp, 2020-02-06:
Can you check if pull-request #14 fixed at least the setup part?

This indeed looks like a duplicate of issue #11 otherwise

NaotoFujihiro, 2020-02-06:
Thank you very much!!
I solved this issue.

What I did is the same thing as #11.

## What I did
```
$ brew unlink md5sha1sum
$ brew install coreutils
```

The reason I ran the command `$ brew unlink md5sha1sum` is I faced the following error while running the command `$ brew install coreutils`.

```
Error: Cannot install coreutils because conflicting formulae are installed.
  md5sha1sum: because both install `md5sum` and `sha1sum` binaries

Please `brew unlink md5sha1sum` before continuing.
```

After running the command `$ brew install coreutils`, the following error didn't occur while running the command `./setup.sh`.

```
tools/gen_key_materials.sh: line 95: [: -gt: unary operator expected
```

## For more info
```
$ bash --version
GNU bash, version 5.0.11(1)-release (x86_64-apple-darwin18.6.0)
Copyright (C) 2019 Free Software Foundation, Inc.
```

closed 2020-02-06 by NaotoFujihiro

NaotoFujihiro, 2020-02-07:
@jmichelp

After I installed `coreutils` and successfully installed TockOS and the OpenSK application, I uninstalled `coreutils` and  at the f5ed8a09f6196d28e137b544740957cda37e2c23, #14 fixed point, ran the command `./setup.sh`.

The result is I installed them successfully without the error `tools/gen_key_materials.sh: line 95: [: -gt: unary operator expected`.

Thanks a lot.