google/OpenSK #24: Cargo audit warns about

gendx, 2020-02-06:
## Expected Behavior

Cargo audit doesn't find any advisory.

## Actual Behavior

[RUSTSEC-2019-0031](https://rustsec.org/advisories/RUSTSEC-2019-0031) is found.

```
Crate:  spin
Title:  spin is no longer actively maintained
Date:   2019-11-21
URL:    https://rustsec.org/advisories/RUSTSEC-2019-0031
Dependency tree: 
spin 0.5.2
├── ring 0.16.11
│   └── crypto 0.1.0
│       └── ctap2 0.1.0
└── linked_list_allocator 0.6.6
    └── libtock 0.1.0
        ├── ctap2 0.1.0
        └── crypto 0.1.0
```

## Next steps

1. I filed an issue for `linked_list_allocator`: https://github.com/phil-opp/linked-list-allocator/issues/22.
  - The libtock-rs version we're based on uses `spin` to lock the allocator ([here](https://github.com/tock/libtock-rs/blob/ab2c945184b98ecae3e70ac678e9f5231deef73b/src/entry_point.rs#L360)), even though there is only one thread of execution so the lock isn't necessary and I don't expect much issue given that only one thread acquires the lock only once.
  - Upstream libtock-rs doesn't lock the allocator anymore (https://github.com/tock/libtock-rs/pull/107), so we can pick that up as a patch in the short term.
2. The `ring` dependency is only used for unit tests, and the advisory is tracked here: https://github.com/briansmith/ring/issues/921.

gendx, 2020-02-19:
The `linked_list_allocator` warning is now removed, due to not depending on the `spin` feature.

kaczmarczyck, 2023-11-09:
After #662, cargo audit is happy.

closed 2023-11-09 by kaczmarczyck