google/OpenSK #25: better handling/docs of missing `rustup`

roycewilliams, 2020-02-07:
## Expected Behavior

- Installer detects that `rustup` is missing and gives appropriate guidance
- Install doc gives brief overview of bootstrapping Rust enough to support deployment

## Actual Behavior

- Install doc assumes specific Rust setup on Linux
- Naive installation produces uncaught errors when trying to execute `rustup`

## Steps to Reproduce the Problem

1. Follow install doc for first-time initialization of a fresh device

## Specifications

- Version: d8c29b7
- Platform: Linux [redacted] 4.15.0-74-generic #84-Ubuntu SMP Thu Dec 19 08:06:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```
$ ./setup.sh 
Submodule 'third_party/libtock-rs' (https://github.com/tock/libtock-rs) registered for path 'third_party/libtock-rs'
Submodule 'third_party/tock' (https://github.com/tock/tock) registered for path 'third_party/tock'
Cloning into '/media/S_DATA/data/linux-root/usr/local/src/sec/security-keys/OpenSK/third_party/libtock-rs'...
Cloning into '/media/S_DATA/data/linux-root/usr/local/src/sec/security-keys/OpenSK/third_party/tock'...
Submodule path 'third_party/libtock-rs': checked out 'ab2c945184b98ecae3e70ac678e9f5231deef73b'
Submodule path 'third_party/tock': checked out '862452b77ae0fc160231a2250de385dc7c358ef7'
[-] Applying patch "01-persistent-storage.patch"... DONE.
[-] Applying patch "02-usb.patch"... DONE.
[-] Applying patch "03-app-memory.patch"... DONE.
[-] Applying patch "04-rtt.patch"... DONE.
[-] Applying patch "01-linked_list_allocator.patch"... DONE.
[-] Applying patch "02-panic_console.patch"... DONE.
[-] Applying patch "03-timer.patch"... DONE.
[-] Applying patch "04-public_syscalls.patch"... DONE.
[-] Applying patch "05-bigger_heap.patch"... DONE.
Signature ok
subject=CN = Google OpenSK CA
Getting Private key
Signature ok
subject=CN = Google OpenSK Hacker Edition
Getting CA Private Key
./setup.sh: line 45: rustup: command not found
Collecting tockloader
  Downloading https://files.pythonhosted.org/packages/19/5f/8dd60c92fc9284afc304dd7a87bfba5e2d8bc0bd49305ea090808dd517b2/tockloader-1.3.1-py3-none-any.whl
Collecting colorama>=0.3.7 (from tockloader)
  Downloading https://files.pythonhosted.org/packages/c9/dc/45cdef1b4d119eb96316b3117e6d5708a08029992b2fee2c143c7a0a5cc5/colorama-0.4.3-py2.py3-none-any.whl
Collecting pytoml>=0.1.11 (from tockloader)
  Downloading https://files.pythonhosted.org/packages/a5/47/c7f8a0f210ad18576840922e0b504f0b7f5f73aea4a52ab14c5b58517edf/pytoml-0.1.21-py2.py3-none-any.whl
Collecting pyserial>=3.0.1 (from tockloader)
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 1.6MB/s 
Collecting crcmod>=1.7 (from tockloader)
  Downloading https://files.pythonhosted.org/packages/6b/b0/e595ce2a2527e169c3bcd6c33d2473c1918e0b7f6826a043ca1245dd4e5b/crcmod-1.7.tar.gz (89kB)
    100% |████████████████████████████████| 92kB 3.6MB/s 
Collecting argcomplete>=1.8.2 (from tockloader)
  Downloading https://files.pythonhosted.org/packages/82/7d/455e149c28c320044cb763c23af375bd77d52baca041f611f5c2b4865cf4/argcomplete-1.11.1-py2.py3-none-any.whl
Collecting importlib-metadata<2,>=0.23; python_version == "3.6" (from argcomplete>=1.8.2->tockloader)
  Downloading https://files.pythonhosted.org/packages/8b/03/a00d504808808912751e64ccf414be53c29cad620e3de2421135fcae3025/importlib_metadata-1.5.0-py2.py3-none-any.whl
Collecting zipp>=0.5 (from importlib-metadata<2,>=0.23; python_version == "3.6"->argcomplete>=1.8.2->tockloader)
  Downloading https://files.pythonhosted.org/packages/be/69/4ac28bf238f287f1677f41392e24d2c4ffafcf11648c23824f5f62ef6ccb/zipp-2.1.0-py3-none-any.whl
Building wheels for collected packages: crcmod
  Running setup.py bdist_wheel for crcmod ... done
  Stored in directory: /home/royce/.cache/pip/wheels/50/24/4d/4580ca4a299f1ad6fd63443e6e584cb21e9a07988e4aa8daac
Successfully built crcmod
Installing collected packages: colorama, pytoml, pyserial, crcmod, zipp, importlib-metadata, argcomplete, tockloader
Successfully installed argcomplete-1.11.1 colorama-0.4.3 crcmod-1.7 importlib-metadata-1.5.0 pyserial-3.4 pytoml-0.1.21 tockloader-1.3.1 zipp-2.1.0
./setup.sh: line 47: rustup: command not found
    Updating crates.io index
warning: spurious network error (2 tries remaining): failed to create temporary file '/home/royce/.cargo/registry/index/github.com-1ecc6299db9ec823/.git/objects/pack/pack_git2_qiKoAB': Permission denied; class=Os (2)
warning: spurious network error (1 tries remaining): failed to create temporary file '/home/royce/.cargo/registry/index/github.com-1ecc6299db9ec823/.git/objects/pack/pack_git2_zaDJgw': Permission denied; class=Os (2)
error: failed to fetch `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to create temporary file '/home/royce/.cargo/registry/index/github.com-1ecc6299db9ec823/.git/objects/pack/pack_git2_NpOVKv': Permission denied; class=Os (2)
royce@mycroft:/usr/local/src/sec/security-keys/OpenSK$ rustup
rustup: command not found

$ board=nrf52840_dongle ./deploy.sh os app

Command elf2tab not found. Have you run the setup.sh script?

```

closed 2020-02-09 by ia0

roycewilliams, 2020-02-10:
Perfect and quick - thanks very much!

For anyone else who happens to run into this, just blow your local repo away and re-clone. Otherwise,  your local OpenSK will be in a nonfunctional state; running ./setup.sh will fail to properly apply patches, and you'll get a jumble of something like this:

```
$ ./setup.sh 
[-] Applying patch "01-persistent-storage.patch"... error: patch failed: boards/nordic/nrf52dk_base/src/lib.rs:105
error: boards/nordic/nrf52dk_base/src/lib.rs: patch does not apply
error: patch failed: chips/nrf52/src/nvmc.rs:3
error: chips/nrf52/src/nvmc.rs: patch does not apply
error: patch failed: kernel/src/callback.rs:50
error: kernel/src/callback.rs: patch does not apply
```
[etc]

If you encounter this, just remove your repo entirely and re-clone it.

(But I expect that very few people will - there was a very small window of time before this fix was applied, and not many are likely to have pulled _and_ didn't already have the right Rust setup)

Thanks again!

ia0, 2020-02-10:
Actually, there was a slightly simpler way. The reset.sh script permits to revert uncommitted changes (including in the submodules) to restore the repository to an almost clean cloned state. I'm adding documentation in #31.

PR #29 (2020-02-09): Check that rustup and pip3 are installed
Fixes #25

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
commit 796261d5 (2020-02-09): Check that rustup and pip3 are installed

Fixes #25