google/OpenSK #27: Document how to check whether the device is working

markushx, 2020-02-08:
## Expected Behavior

Documentation should explain what to expect after the device has been flashed. Is there supposed to be a ttyUSB? How do you check whether the device works properly (https://webauthn.io/)?

## Actual Behavior

Only setup, compiling, flashing is documented.

## Steps to Reproduce the Problem

1. Follow documentation

superderon, 2020-02-11:
In my case, i checked it by going to [webauthn](https://webauthn.io/), don't forget to push reset the board (mine is the DK version), insert my name, then try Register. It will ask for your secure device and touch any button available (in my case there are 4 buttons, any of these will work). After that your name will be registered. Then you can login with your name and authenticate again.

Or you can just watch this [video](https://www.youtube.com/watch?v=XEmEADLu8yk)

kaczmarczyck, 2020-02-11:
Can you please clarify your question or confirm? My understanding is: You would like us to mention webauthn.io in our documentation as the recommended end-to-end test after flashing OpenSK? Or is this asking for other ways to test your security key?

markushx, 2020-02-11:
Yes, I think the documentation should mention how the device would appear in dmesg, lsusb and/or in /dev/tty* as well as a basic test of the security functionality with webauthn.io.

PR #39 (2020-02-19): Add troubleshooting instructions.
Fixes #27

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-02-19 by jmichelp