google/OpenSK #28: Error while authenticating on Windows 10 when using userVerification

scottbrady91, 2020-02-09:
## Expected Behavior
When userVerification set to preferred or required: User enters their PIN and the presses a button. User should then be authenticated.

## Actual Behavior
User enters their PIN and presses a button. User is then stuck on "Touch your authenticator" Windows Hello screen.

See attached Windows logs for errors: [OpenSK UserVerification Error.zip](https://github.com/google/OpenSK/files/4176596/OpenSK.UserVerification.Error.zip)

## Steps to Reproduce the Problem

1. Register new key
1. Authenticate using key with userVerification set to preferred or required

## Specifications

- Platform: Windows 10

jmichelp, 2020-02-11:
Can you tell us on which website you were doing the test?

scottbrady91, 2020-02-12:
I've tested on https://webauthn.me/ and https://fido.identityserver.com/.

Although, the error looks like it is happening between the authenticator and WebAuthn, so I would be surprised if the FIDO RP or WebAuthn usage would be to blame.

jmichelp, 2020-02-20:
Ok, we will try to reproduce that behavior so that we can debug it and fix it.

As you were mentioning Windows Hello, I wanted to ensure you were not trying to log in into your workstation using Windows Hello. This requires a CTAP2 extension called hmac-secret that we haven't developed yet.

abhvious, 2020-02-24:
I gave all students in my CY2550 section a Nordic with OpenSK flashed.  We can reliably repro this on Windows.  To add some more info, the registration process works fine, but requires the user to enter a PIN for the token.  When the user tries to login, Windows Security kicks in and asks for the PIN.  We can enter it, but the rest of the flow doesn't engage. The window just sits there and then times out.  The problem is between the browser and the token, because the blue lights on the token are not engaged.

The token works fine on MacOs, on Linux, and on Linux-on-virtualbox.  We are using this against an internal site (that I can share with you @jmichelp), but the same happens on webauthn.io.

BTW, I just wanted to also applaud you for this project, it is REALLY great for the community and for learning.

kaczmarczyck, 2020-03-13:
I recorded this log on a Windows 10 with my dev board on commit dabbe3860570937e6fc35f31f6b5a13c4e401d12.
[windows10_uv_log.txt](https://github.com/google/OpenSK/files/4330332/windows10_uv_log.txt)

Setting the PIN fails once, probably debug logging is too slow. It works fine without debug logging. With out without, when I try Login, it fails the same way: The authenticator responds properly, but Windows waits forever.

kaczmarczyck, 2020-03-13:
Last log was from fido.identityserver.com, this one is from webauthn.me (PIN already set, so this log is a bit shorter and more readable).
[windows10_uv_log_webauthn_me.txt](https://github.com/google/OpenSK/files/4330364/windows10_uv_log_webauthn_me.txt)

Windows keeps the dialog open: "Take action on your security key."

kaczmarczyck, 2020-03-13:
The Yubikey works for the same sequence of steps.

closed 2020-08-17 by kaczmarczyck

kaczmarczyck, 2020-08-17:
@scottbrady91 Did this solve your issue? Please reopen if we missed anything.

PR #143 (2020-08-17): Do not generate an empty PublicKeyCredentialUserEntity in response to…
Do not generate an empty PublicKeyCredentialUserEntity in response to GetAssertion

Fixes #28 
Fixes #114 

The current version of Windows WebAuthn does not seem to support an empty PublicKeyCredentialUserEntity.

I tested it in the following environments
- windows10 pro 64bit 1909
- Non Resident Key Resister and Authenticate
- https://yubionfido2demo.yubion.com/register_jp.html
- https://fido.identityserver.com
- https://webauthn.io/

Before the modification, the user confirmation button did not respond during the authenti [... cut]