google/OpenSK #30: JLinkExe missing during flash build (nrf52840_dongle)

roycewilliams, 2020-02-10:
I'm the "total noob use case" smoke tester this week. :laughing:

## Expected Behavior

* Missing JLinkExe detected and cleanly handled 
* Note of JLinkExe as a dependency in docs
* Doc hints (maybe to the Tock Getting Started doc?)

## Actual Behavior

* Uncaught error if JLinkExe is missing

## Steps to Reproduce the Problem

* Run `board=nrf52840_dongle ./deploy.sh os app` without JLinkExe available

## Specifications

- Version: 796261d
- Platform: Linux [redacted] 4.15.0-74-generic #84-Ubuntu SMP Thu Dec 19 08:06:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

[...]
```
tockloader  flash --address 0x00000 --jlink --board nrf52dk target/thumbv7em-none-eabi/release/nrf52840_dongle.bin
Flashing binar(y|ies) to board...
Using known arch and jtag-device for known board nrf52dk
Traceback (most recent call last):
  File "/home/royce/.local/bin/tockloader", line 11, in <module>
    sys.exit(main())
  File "/home/royce/.local/lib/python3.6/site-packages/tockloader/main.py", line 593, in main
    args.func(args)
  File "/home/royce/.local/lib/python3.6/site-packages/tockloader/main.py", line 216, in command_flash
    tock_loader.flash_binary(binary, args.address)
  File "/home/royce/.local/lib/python3.6/site-packages/tockloader/tockloader.py", line 63, in flash_binary
    self.channel.flash_binary(address, binary)
  File "/home/royce/.local/lib/python3.6/site-packages/tockloader/jlinkexe.py", line 96, in flash_binary
    self._run_jtag_commands(commands, binary)
  File "/home/royce/.local/lib/python3.6/site-packages/tockloader/jlinkexe.py", line 63, in _run_jtag_commands
    p = subprocess.run(jlink_command.split(), stdout=subprocess.PIPE, stderr=subprocess.PIPE)
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'JLinkExe': 'JLinkExe'

Makefile:24: recipe for target 'flash' failed
make: *** [flash] Error 1
make: Leaving directory '/media/S_DATA/data/linux-root/usr/local/src/sec/security-keys/OpenSK/third_party/tock/boards/nordic/nrf52840_dongle'
```

roycewilliams, 2020-02-10:
Looking more closely, I see that the upstream Tock docs do cover the docs part somewhat. If it is only possible to use the [J-Link EDU](https://shop-us.segger.com/J_Link_EDU_p/8.08.90.htm) with the full development board, and not the USB stick version, cover that in the README would also be helpful for users who don't have a lot of JTAG experience. :flushed:

conashin, 2020-02-11:
So, do we need any other things when flashing nrf52840 dongle?

superderon, 2020-02-11:
> Looking more closely, I see that the upstream Tock docs do cover the docs part somewhat. If it is only possible to use the [J-Link EDU](https://shop-us.segger.com/J_Link_EDU_p/8.08.90.htm) with the full development board, and not the USB stick version, or what combinations might work, covering that in the README would also be helpful for users who don't have a lot of JTAG experience.

Do you already have Jlink installed?
If not then install from this .deb
[https://www.segger.com/downloads/jlink/JLink_Linux_x86_64.deb](JLink Linux 64)

conashin, 2020-02-11:
> Do you already have Jlink installed?
> If not then install from this .deb
> [Jlink Linux 64](https://www.segger.com/downloads/jlink/JLink_Linux_x86_64.deb)

Can I install a jlink flasher?
[https://www.segger.com/downloads/flasher/JLink_Linux_x86_64.deb](url)

superderon, 2020-02-11:
@conashin if you had an error like
`FileNotFoundError: [Errno 2] No such file or directory: 'JLinkExe': 'JLinkExe'`
Then yes, but make sure you're downloading the right one for your platform.
[JLink Download Page](https://www.segger.com/downloads/jlink/#J-LinkSoftwareAndDocumentationPack)

conashin, 2020-02-12:
@superderon System pop-up this window when I was trying to flash it.
[image]

Does it need something else with hardware?

jmichelp, 2020-02-12:
According to the very first line in the dialog window, your JLink probe is not detected: "No emulators connected via USB"

jrvanwhy, 2020-02-13:
I deployed OpenSK to a nrf52840-DK using OpenOCD rather than JLinkExe. To do so, I needed an up-to-date version of OpenOCD (newer than my distro packages), and I made tweaks to tockloader, tock, and OpenSK's deploy.sh.

I'll work on upstreaming (less hacky versions of) the changes I've made. Upstreaming will take some time, as I expect to make changes serially across three repositories. I'll open a separate issue for OpenOCD support when I've made enough progress to put together a coherent issue title/body.

jmichelp, 2020-02-13:
Thanks Johnathan! Be aware that I'm currently almost done rewriting the `deploy.sh` script in python3. This should give us more flexibility and portability (python3 is already a dependency for tockloader anyway).

Let's coordinate so that I'm not adding more work to merge your contributions.

jrvanwhy, 2020-02-13:
> Thanks Johnathan! Be aware that I'm currently almost done rewriting the `deploy.sh` script in python3. This should give us more flexibility and portability (python3 is already a dependency for tockloader anyway).
> 
> Let's coordinate so that I'm not adding more work to merge your contributions.

I opened #35 to track/coordinate this effort. I don't expect it to collide with your deploy.sh changes. I added a flag to tockloader (https://github.com/tock/tockloader/pull/52) then tweaked deploy.sh and tock so that I could pass the flags through.

jmichelp, 2020-02-20:
PR #40 now checks for `JLinkExe`.

Thanks to Johnathan and the community, it seems that we're going to be able to support different flashing methods, including using the pre-flashed Nordic bootloader. Covering how to setup the JTAG probes on everyone's environment is in my opinion out of scope for this project.

closed 2020-02-20 by jmichelp

roycewilliams, 2020-02-20:
Yep, works just as intended - thanks for this!

For the upcoming native Nordic bootloader work - is that tracking under #35 ? Which issue(s) should I follow to track that?