google/OpenSK #33: Build successfully but not recognized in webauth or U2F

garryleee, 2020-02-11:
Hi I am the newbie and get a new nRF52840DK board and get clone of OpenSK repository, following installation guide with all the required software installed e.g. rust, tockloader etc and flash successfully.

The question is it seems everything is flushed including the original bootloader when I config as USB HID with nRF52840DK originally. Also when I use the board as USB, use web server with chrome requested to insert security key in USB port (WebAthun) or Yubico testing U2F website, I plug my board as USB peripheral and it does not recognize.

Can you please advice me where goes wrong?
Thank you! ChillyBean

superderon, 2020-02-11:
Did you already tried to push the reset button before the web server request authentication?

chillynoodles, 2020-02-11:
> Did you already tried to push the reset button before the web server request authentication?

Yes I did press the reset...:(

superderon, 2020-02-11:
@chillynoodles did the led1-4 blinking when the web asks for authentication?

chillynoodles, 2020-02-11:
> @chillynoodles did the led1-4 blinking when the web asks for authentication?

Noop - I notice the led1-4 light off after flash OpenSK there. I did try again to run and flash again and shows:

Preparing to uninstall apps...
Only one app on board. Uninstalling ctap2
Uninstall complete.
No apps on board.
Finished in 1.976 seconds

    Finished release [optimized] target(s) in 0.02s
Flashing binar(y|ies) to board...
Finished in 0.561 seconds

Installing apps on the board...
Finished in 3.654 seconds
and the board LED lights all go off - that's weird why I think it is flushed all....

Thank you for your help

superderon, 2020-02-11:
@chillynoodles in my case, the led is off all the time, only blinks when the web ask for authentication.

Your log seems fine.

Have you tried this?
`sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/`
`sudo udevadm control --reload`

chillynoodles, 2020-02-11:
> @chillynoodles in my case, the led is off all the time, only blinks when the web ask for authentication.
> 
> Your log seems fine.
> 
> Have you tried this?
> `sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/`
> `sudo udevadm control --reload`

Yes - i do. Can I ask you a question on your board behavior please?
When you press reset button of the board when connected with JLink, then unplug and use the web ask for auth with chrome, plug the board (USB peripherals) and it is passed?

I actually use https://demo.yubico.com/u2f/ and it prompts with "insert your security key", and I plug with board USB peripheral port and nothing happened...
Did you do the same steps?

Thanks,

superderon, 2020-02-11:
> Yes - i do. Can I ask you a question on your board behavior please?
> When you press reset button of the board when connected with JLink, then unplug and use the web ask for auth with chrome, plug the [...]

Here is my steps, 
- Flash the app using
`$ board=nrf52840dk ./deploy.sh app os`
- Remove the JTag USB
- Connect the board into host PC via USB peripheral port (near reset button)
- Press reset (important)
- Open [https://demo.yubico.com/u2f](https://demo.yubico.com/u2f/)
- The web asks for authentication then my board's Led 1-4 blinking, just press any button from 1-4 and its authenticated.

By the way, i use Chrome.

chillynoodles, 2020-02-11:
> Here is my steps,
>
> * Flash the app using
>   `$ board=nrf52840dk ./deploy.sh app os`
> * Remove the JTag USB
> * Connect the board into host PC via USB peripheral port (near reset button)
> * Press reset [...]

I redo with what you did and press the reset button - it works!!!! 
Really appreciate for your help!!! :)

chillynoodles, 2020-02-17:
Ooops - I can config and run fine with my nrf52840dk board and when I moved to nrf52840_dongle, run with existed same code by:

 

- Flash the app using $ board=nrf52840_dongle ./deploy.sh app os

- Remove the JTag USB, NRF52840 dongle and open https://demo.yubico.com/u2f

- The web asks for authentication and plug nrf52840 dongle and it does not recognize and nothing happend...
Anyone can help to advice why it's working in nrf52840dk board but not with dongle please? 

$ board=nrf52840_dongle ./deploy.sh app os
info: syncing channel updates for 'nightly-2020-02-03-x86_64-unknown-linux-gnu'

  nightly-2020-02-03-x86_64-unknown-linux-gnu unchanged - rustc 1.42.0-nightly (f43c34a13 2020-02-02)

info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
make: Entering directory '/home/chillybean/OpenSK/third_party/tock/boards/nordic/nrf52840_dongle'
    Finished release [optimized + debuginfo] target(s) in 0.01s
   text	   data	    bss	    dec	    hex	filename
 114176	   1708	 260436	 376320	  5be00	target/thumbv7em-none-eabi/release/nrf52840_dongle
tockloader  flash --address 0x00000 --jlink --board nrf52dk target/thumbv7em-none-eabi/release/nrf52840_dongle.bin
Flashing binar(y|ies) to board...
Using known arch and jtag-device for known board nrf52dk
Finished in 7.898 seconds

make: Leaving directory '/home/chillybean/OpenSK/third_party/tock/boards/nordic/nrf52840_dongle'
Preparing to uninstall apps...
No apps are installed on the board

    Finished release [optimized] target(s) in 0.02s
Flashing binar(y|ies) to board...
Finished in 7.862 seconds

Installing apps on the board...
Finished in 22.409 seconds

closed 2020-02-20 by chillynoodles