google/OpenSK #35: Add OpenOCD support

jrvanwhy, 2020-02-13:
OpenSK uses `tockloader` to deploy the kernel and apps, and currently passes flags that cause `tockloader` to invoke `JLinkExe`. We should allow users to use OpenOCD instead of `JLinkExe`, by making the flags passed to `tockloader` configurable/changable.

Additionally, because distributions often have an outdated version of OpenOCD that cannot deploy to a nrf52840-DK, OpenSK should allow users to use a custom version of OpenOCD. I've already opened [tock/tockloader/pull/52](https://github.com/tock/tockloader/pull/52) to enable `tockloader` to invoke a custom version of OpenOCD.

jrvanwhy, 2020-02-25:
The `tockloader` PR (tock/tockloader#52) has been merged. I opened the following issue to get support for OpenOCD arguments added to tock: https://github.com/tock/tock/issues/1626

kaczmarczyck, 2026-04-24:
This would need to be supported in [Wasefire](https://github.com/google/wasefire) with #768, and then we would inherit it automatically. Please open an issue there if you are still interested.

closed 2026-04-24 by kaczmarczyck