google/OpenSK #37: error: failed to select a version for `libtock.

xaqfan, 2020-02-18:
I'm trying to download dongle, but I made a mistake in the middle.
I have downloaded the resource files under OpenSK-master/third_party
I have no ideas.

board=nrf52840_dongle ./deploy.sh os app
nightly-2020-02-03-i686-unknown-linux-gnu unchanged - rustc 1.42.0-nightly (f43c34a13 2020-02-02)

info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
make: Entering directory '/home/xaqfan/Tock_OS/OpenSK-master/third_party/tock/boards/nordic/nrf52840_dongle'
    Finished release [optimized + debuginfo] target(s) in 0.01s
   text	   data	    bss	    dec	    hex	filename
 102912	   1696	 260448	 365056	  59200	target/thumbv7em-none-eabi/release/nrf52840_dongle
tockloader  flash --address 0x00000 --jlink --board nrf52dk target/thumbv7em-none-eabi/release/nrf52840_dongle.bin
Flashing binar(y|ies) to board...
Using known arch and jtag-device for known board nrf52dk
Finished in 7.745 seconds

make: Leaving directory '/home/xaqfan/Tock_OS/OpenSK-master/third_party/tock/boards/nordic/nrf52840_dongle'
Preparing to uninstall apps...
No apps are installed on the board

    Updating crates.io index
error: failed to select a version for `libtock`.
    ... required by package `ctap2 v0.1.0 (/home/xaqfan/Tock_OS/OpenSK-master)`
versions that meet the requirements `*` are: 0.1.0

the package `ctap2` depends on `libtock`, with features: `panic_console` but `libtock` does not have these features.

failed to select a version for `libtock` which could resolve this conflict

Would someone can help me ? 
Thanks!

jmichelp, 2020-02-18:
This looks like as if the patches were not applied.

Try the following command: `./reset.sh; ./setup.sh`

Then you should be able to run `board=nrf52840_dongle ./deploy.sh os app`

xaqfan, 2020-02-19:
I have decompressed OpenSK-master.zip again ,and setup.sh  ,board=nrf52840_dongle ./deploy.sh os app
It will still  report the same mistake.
./setup.sh 
[-] Applying patch "01-persistent-storage.patch"... DONE.
[-] Applying patch "02-usb.patch"... DONE.
[-] Applying patch "03-app-memory.patch"... DONE.
[-] Applying patch "04-rtt.patch"... DONE.
[-] Applying patch "01-linked_list_allocator.patch"... DONE.
[-] Applying patch "02-panic_console.patch"... DONE.
[-] Applying patch "03-timer.patch"... DONE.
[-] Applying patch "04-public_syscalls.patch"... DONE.
[-] Applying patch "05-bigger_heap.patch"... DONE.
Signature ok
subject=CN = Google OpenSK CA
Getting Private key
Signature ok
subject=CN = Google OpenSK Hacker Edition
Getting CA Private Key
info: syncing channel updates for 'nightly-2020-02-03-i686-unknown-linux-gnu'

  nightly-2020-02-03-i686-unknown-linux-gnu unchanged - rustc 1.42.0-nightly (f43c34a13 2020-02-02)

info: checking for self-updates
Requirement already up-to-date: tockloader in /home/xaqfan/.local/lib/python3.5/site-packages (1.3.1)
Requirement already satisfied, skipping upgrade: pyserial>=3.0.1 in /home/xaqfan/.local/lib/python3.5/site-packages (from tockloader) (3.4)
Requirement already satisfied, skipping upgrade: argcomplete>=1.8.2 in /home/xaqfan/.local/lib/python3.5/site-packages (from tockloader) (1.11.1)
Requirement already satisfied, skipping upgrade: crcmod>=1.7 in /home/xaqfan/.local/lib/python3.5/site-packages (from tockloader) (1.7)
Requirement already satisfied, skipping upgrade: pytoml>=0.1.11 in /home/xaqfan/.local/lib/python3.5/site-packages (from tockloader) (0.1.21)
Requirement already satisfied, skipping upgrade: colorama>=0.3.7 in /home/xaqfan/.local/lib/python3.5/site-packages (from tockloader) (0.4.3)
Requirement already satisfied, skipping upgrade: importlib-metadata<2,>=0.23; python_version == "3.5" in /home/xaqfan/.local/lib/python3.5/site-packages (from argcomplete>=1.8.2->tockloader) (1.5.0)
Requirement already satisfied, skipping upgrade: zipp>=0.5 in /home/xaqfan/.local/lib/python3.5/site-packages (from importlib-metadata<2,>=0.23; python_version == "3.5"->argcomplete>=1.8.2->tockloader) (1.1.0)
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
    Updating crates.io index
     Ignored package `elf2tab v0.4.0` is already installed, use --force to override
xaqfan@xaqfan:~/Tock_OS/OpenSK-master$ board=nrf52840_dongle ./deploy.sh os app
info: syncing channel updates for 'nightly-2020-02-03-i686-unknown-linux-gnu'

  nightly-2020-02-03-i686-unknown-linux-gnu unchanged - rustc 1.42.0-nightly (f43c34a13 2020-02-02)

info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
make: Entering directory '/home/xaqfan/Tock_OS/OpenSK-master/third_party/tock/boards/nordic/nrf52840_dongle'
    Finished release [optimized + debuginfo] target(s) in 0.02s
   text	   data	    bss	    dec	    hex	filename
 102912	   1696	 260448	 365056	  59200	target/thumbv7em-none-eabi/release/nrf52840_dongle
tockloader  flash --address 0x00000 --jlink --board nrf52dk target/thumbv7em-none-eabi/release/nrf52840_dongle.bin
Flashing binar(y|ies) to board...
Using known arch and jtag-device for known board nrf52dk
Finished in 7.895 seconds

make: Leaving directory '/home/xaqfan/Tock_OS/OpenSK-master/third_party/tock/boards/nordic/nrf52840_dongle'
Preparing to uninstall apps...
No apps are installed on the board

    Updating crates.io index
error: failed to select a version for `libtock`.
    ... required by package `ctap2 v0.1.0 (/home/xaqfan/Tock_OS/OpenSK-master)`
versions that meet the requirements `*` are: 0.1.0

the package `ctap2` depends on `libtock`, with features: `panic_console` but `libtock` does not have these features.

failed to select a version for `libtock` which could resolve this conflict

gendx, 2020-02-19:
In your output, I can see:
```
make: Entering directory '/home/xaqfan/Tock_OS/OpenSK-master/third_party/tock/boards/nordic/nrf52840_dongle'
Finished release [optimized + debuginfo] target(s) in 0.02s
```

The fact that compilation only took 0.02 seconds seems to indicate that there was already a compiled version of Tock in your folder, which contradicts a "fresh" clone of OpenSK.

---

The following error message:

> The package ctap2 depends on libtock, with features: panic_console but libtock does not have these features.

is in contradiction with:
```
[-] Applying patch "02-panic_console.patch"... DONE.
```

---

Could you try `./reset.sh` before the `./setup.sh`? Or doing a fresh `git clone https://github.com/google/OpenSK`?

jmichelp, 2020-02-19:
Also, downloading a ZIP file from GitHub shouldn't work because you won't have the git metadata that is needed to bootstrap `tock` and `libtock-rs` submodules and therefore the `setup.sh` script should fail, as shown below:

```shell
$ unzip OpenSK-master.zip
Archive:  OpenSK-master.zip
51f201653c3054d90cfe6d535a260935f76aba45
[...]
$ cd OpenSK-master
$ ./setup.sh
fatal: not a git repository
```

xaqfan, 2020-02-20:
Thank you very much!
that method is useful。But I have other problems after git clone the newest latest version.

This issue can close!

closed 2020-02-20 by xaqfan

adamberglund, 2020-02-20:
UNSUBSCRIBE

 �

 �