google/OpenSK #42: Deploy.py Requires Tockloader 1.4.0, which is not yet released via pip

redsky17, 2020-02-19:
## Expected Behavior

The latest version of tockloader available via pip install is tockloader 1.3.1: https://pypi.org/project/tockloader/

Running the setup.sh script therefore installs tockloader 1.3.1 instead of a more recent version.  Based on the commit history in the tockloader repository, the 'sticky' argument of the install function isn't available in 1.3.1 (released Nov. 2018): https://github.com/tock/tockloader/blob/v1.3.1/tockloader/tockloader.py#L86

It appears that this 'sticky' argument was added much more recently (Oct. 2019): 
https://github.com/tock/tockloader/commit/22f34653559dcc9a81690183e9cbf7e3eef4ba69

This causes the deploy.py script that was recently released to fail due to an unexpected argument in the call to tock.install() (here's the log excerpt):

```
info: Installing Tock application ctap2
Traceback (most recent call last):
  File "./deploy.py", line 251, in install_elf_file
    tock.install(tabs, replace="yes",
TypeError: install() got an unexpected keyword argument 'sticky'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./deploy.py", line 430, in <module>
    main(parser.parse_args())
  File "./deploy.py", line 318, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 305, in run
    self.build_and_install_opensk()
  File "./deploy.py", line 199, in build_and_install_opensk
    self.install_elf_file(os.path.join(
  File "./deploy.py", line 253, in install_elf_file
    except tockloader.exceptions.TockLoaderException as e:
AttributeError: module 'tockloader.tockloader' has no attribute 'exceptions'
```
This causes the deployment of the OpenSK application to fail.  I'm not sure if there are other differences between tockloader 1.3.1 and the version that OpenSK is expecting (based on the logs in https://github.com/google/OpenSK/blob/master/docs/install.md, it appears to be a 1.4.0-dev version).  It might be enough to just remove the 'sticky' argument, but there may be other breaking changes as well.

## Steps to Reproduce the Problem

1. Run setup.sh - This reports that tockloader 1.3.1 is up-to-date already:
    `Requirement already up-to-date: tockloader in *omitted*/python3.8/site-packages (1.3.1)`
1. Run deploy.py os --board=nrf52840_dk
1. Run deploy.py app --opensk (this is where it fails)

xaqfan, 2020-02-20:
You need tockloader-master version.

jmichelp, 2020-02-20:
We don't need the `sticky` parameter as the default value is the one we use.
I'll remove it. Thanks for reporting.

PR #45 (2020-02-20): Improve deploy script.
- Check that JLinkExe is properly installed
- Verify that the app is installed
- Always report failures in the exit code
- Add vscode settings for Python code formatting using Google style
- Fix issue #42 regarding tockloader version and sticky parameter

Fixes #42 

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-02-20 by jmichelp