google/OpenSK #44: Can OpenSK be used on RiscV?

bling206, 2020-02-20:
Hi 

I found that Rust could be built on RiscV (https://github.com/almindor/riscv-rust-quickstart). 
Does it mean that OpenSK can also be built and used on RiscV? 
Thank you

jmichelp, 2020-02-20:
Tock already has support for RISC-V too.

But at the moment we are not aware of an existing RISC-V board that meets all the following criteria:
* commercially available so that we can easily get one and test
* supported by Tock OS upstream (see https://github.com/tock/tock/blob/master/boards/README.md)
* have working USB controller, a button and at least 1 LED

bling206, 2020-02-24:
Hi Jmichelp

Thanks for replying. 
What is the requirement for supporting Tock OS?
I just get a RISC-V board which has USB controller , a button and more than 1 LED.
But it is not on the Tock OS support list.
How do I know if this board can be used for Tock OS.

Thank you!!

gendx, 2020-02-24:
> What is the requirement for supporting Tock OS?
> I just get a RISC-V board which has USB controller , a button and more than 1 LED.
> But it is not on the Tock OS support list.
> How do I know if this board can be used for Tock OS.

The best would be to ask on the Tock OS repository or mailing list.

In general, adding support for a board on Tock OS requires:
- A suitable CPU arch - currently Cortex-M variants and RISC-V (`rv32i`) are already supported: https://github.com/tock/tock/tree/master/arch
- A suitable [chip](https://github.com/tock/tock/tree/master/chips). If your board's chip is already there great! Otherwise you'd typically need to implement that, at least in terms of GPIOs (button, LED), USB controller, as well as timers and a random number generator (also needed for OpenSK). I'd add that some kind of UART implementation is almost mandatory as well to be able to debug while you're developing all that.
- A [board](https://github.com/tock/tock/tree/master/boards), to combine all the pieces together.
- Additionally you might have to update some of the [capsules](https://github.com/tock/tock/tree/master/capsules).

bling206, 2020-02-24:
Hi  Gendx

Thanks for replying.
I think my RISC-V board meets the requirements which you mentioned about.
I will try to see if I could make it work although I am a beginner of Rust.
Thank you a lot.

gendx, 2020-02-24:
> I think my RISC-V board meets the requirements which you mentioned about.
> I will try to see if I could make it work although I am a beginner of Rust.

Sounds great :) Let us know about your progress!

Some more pointers I can recommend to get started with Rust and Tock:
- [Rust by example](https://doc.rust-lang.org/rust-by-example/).
- Getting started to [use Tock](https://github.com/tock/tock#use-tock).
- The [Tock mailing list](https://groups.google.com/forum/#!forum/tock-dev).
- [Developing Tock OS](https://github.com/tock/tock#develop-tock) itself.
- [Tock documentation](https://github.com/tock/tock/blob/master/doc/README.md) generally.

closed 2020-02-25 by bling206