google/OpenSK #47: Jtagging nRF52840 Dongle

santhoshn, 2020-02-21:
I bought the nRF52840 Dongle and then came to know that extra hardware is needed to load the firmware. Before I try to procure them, I want know if the firmware can be loaded from the USB?

Can this J-Link EDU Mini be used? This one appears economical for a hobbyist:
https://www.segger.com/products/debug-probes/j-link/models/j-link-edu-mini/

jmichelp, 2020-02-21:
I don't know about the J-Link EDU Mini but the [J-Link EDU](https://www.segger.com/products/debug-probes/j-link/models/j-link-edu/) model should definitely work.

With the EDU mini, the connector is a fine pitch one, so you'll need to either use a breakout cable and temporarily solder wire on the castellated holes of the dongle (you only need 4 wires) or use a different cable from TagConnect but none of their cables are cheap. I think the [TC2050-IDC-050](https://www.tag-connect.com/product/tc2050-idc-050) should do the job. Be careful to **not** pick the [TC2050-IDC-050-ALL](http://www.tag-connect.com/product/tc2050-idc-050-all) cable because it will connect pins 5 and 9 and this will create a shortcut on the USB port (pin 5 is GND on the J-Link EDU mini but it's connected to VBUS on the Nordic dongle!).

Regarding flashing directly over USB, we have a contributor working on this. But it's not supported yet.

florian-f, 2020-03-03:
I was in a similar situation, having purchased the nRF52840 Dongle before reading the installation guide. I wasn't aware of the J-Link EDU and decided to purchase the Nordic nRF52840-DK, which has an on-board J-Link.

Now I'm wondering if I can use the on-board J-Link module of the nRF52840-DK to load the firmware on the nRF52840 Dongle.

It looks like it should be possible:

> The nRF52840 DK (Development Kit) board supports programming and debugging external boards with nRF51 or nRF52 System on Chip (SoC)s. To debug an external board with SEGGER J-Link OB IF, connect to the Debug out connector (P19) with a 10-pin cable.

https://infocenter.nordicsemi.com/topic/ug_nrf52840_dk/UG/nrf52840_DK/hw_debug_out.html

jmichelp, 2020-06-29:
This is now obsolete as OpenSK can be flashed directly over USB without external hardware.

closed 2020-06-29 by jmichelp