google/OpenSK #50: Ensure that the deploy script is also covered by Travis runs

jmichelp, 2020-02-21:
## Expected Behavior

The `deploy.py` script should be covered by Travis CI to catch errors.

This will imply several things:
- add a `--dry-run` parameter to the script to run all the actions except execution of `JLinkExe`
- split compilation and flash operations (also useful to support more flashing methods such as USB bootloader, OpenOCD, etc.)
- assess if `tockloader` also has such a dry-run flag
- modify Travis configuration to run deploy script with the different parameters

## Actual Behavior

Script is not run.

## Steps to Reproduce the Problem

N/A

## Specifications

- Version: N/A
- Platform: N/A

jmichelp, 2020-02-21:
As suggested during the review of PR #49, we should also:

- Add linter for python and markdown
- Check for proper formatting of python and markdown files

gendx, 2020-06-22:
The `workflows/boards_build.yml` already tests `./deploy.py --board=<board> --no-app --programmer=none`. I guess the `--programmer=none` corresponds to the `--dry-run` parameter proposed in this issue.

Should we add anything else?
I'd say at the very least not only testing `--no-app`.

kaczmarczyck, 2023-11-09:
Several `--programmer=none` are added.

PR #661 (2023-11-08): Reworks workflows into script
Advantages are:
- Only one set of tests needs maintenance.
- Local results match workflows, no surprises.
- Reduced reliance on GitHub actions.

Fixes #50, #168, #169, #171, #507
closed 2023-11-09 by kaczmarczyck