google/OpenSK #52: Using SWD mode to connect to nRF52840 Dongle?

LiyuCode, 2020-02-23:
In the [doc](https://github.com/LiyuCode/OpenSK/blob/master/docs/install.md#nordic-nrf52840-dongle), the cable conenct to to 10 pads. What are the PINs/pads used to connect to the dongle?

If we use other debug connector rather than Segger J-Link(JTAG mode), can we use the `SWDIO`, `SWDCLK` and `GND`, i.e., `SWD` mode ?

jmichelp, 2020-02-24:
Only 4 out of the 10 pads are really needed to program the dongle, which corresponds to the SWD pins: `SWDIO`, `SWDCLK`, `VDD`, and `GND`. That's how the dongle is being programmed.

The required signals are also available on the castellated holes of the board. `SWDIO` and `SWDCLK` are located very close to the USB port, `GND` signal is available at 3 different places. If your SWD programmer needs a voltage reference, pay attention to use `VDD OUT` castellated hole for that purpose. And you will need to power the board by providing 5V either through the USB port (easiest way) or by applying 5V to the castellated hole labelled `VBUS`.

All the holes on the dongle are spaces with a standard 0.1" pitch (2,54mm) which make it compatible with a bread board. So even though soldering will give you a more reliable connection, I guess you can also do it by maintaining pressure between a header and the dongle to build a programming pod.

abhvious, 2020-02-26:
Indeed, we ended up programming the boards using only the 4 pins and a simple jig built with a breadboard and some headers.  We programmed 75 of these using the Segger edu debugger.

closed 2020-03-11 by jmichelp