google/OpenSK #54: Improve continuous integration

jmichelp, 2020-02-25:
## Expected Behavior

All formatting, tests, and builds are run through Github workflows.

We should also use the [Problem matchers](https://github.com/actions/toolkit/blob/master/docs/problem-matchers.md) wherever we can to have more automated actionable content in the Github PR review system.

## Actual Behavior

Continuous integration runs on Travis-CI

## Steps to Reproduce the Problem

N/A

## Specifications

- Version: N/A
- Platform: N/A

jmichelp, 2020-02-26:
PR #53 started duplicating a few tests from Travis-CI.

jmichelp, 2020-02-28:
Now with PR #61 we have even better coverage than what Travis was doing, tests run in parallel instead of sequentially.

Leaving the bug open though because we're missing clippy due to libtock which triggers clippy errors. Will revisit when we'll be able to bump libtock-rs version.

Also leaving the travis-ci at the moment to ensure that both are catching errors on upcoming PRs and we can disable it once we're confident that everything runs fine.

gendx, 2020-06-22:
Travis-CI has been broken in the last pull requests. Is it still useful or should we just get rid of it?

jmichelp, 2020-06-29:
I removed it. We were only keeping it for a while to see if Github workflows were correctly on par with Travis-CI

PR #61 (2020-03-02): Run everything that Travis-CI is running as Github workflows
Fixes #54 

All tests that are currently being handled by Travis-CI are now ported as workflows.
Additionally we have python formatting check with yapf and python linter with custom ProblemMatchers to better surface the issues.
Beyond scheduled run, also run a cargo audit on pull-request because they may introduce dependencies.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-10-12 by jmichelp