google/OpenSK #55: How to debug, set breakpoints and view console breakpoints?

superderon, 2020-02-26:
Hi, I am new to JLink and Nordic, How do you guys did the debug?
Are you guys doing it on VSCode? I need some guide on debugging this.

gendx, 2020-02-26:
To answer on the console output, you can use the Segger RTT protocol on the nRF52840-DK as explained [here](https://github.com/tock/tock/blob/d52a619e640a5c8cdc9deb74232d145f66fade2d/capsules/src/segger_rtt.rs#L8-L19). We have a [patch](https://github.com/google/OpenSK/blob/master/patches/tock/04-rtt.patch) that enables this console output, as well as a pending pull-request to support it upstream (https://github.com/tock/tock/pull/1556).

You can use the `deploy.py` script with `--debug` to compile OpenSK with more debug output enabled that will be printed on this console.

In general, you can use the `Console` driver of Tock to print your own messages in userspace applications, or the `debug!` macro in the kernel.

closed 2020-02-28 by superderon

gendx, 2020-02-28:
Also, note that there is a bug in Tock affecting debugging, but a fix is on the way in https://github.com/google/OpenSK/pull/60 (see also https://github.com/tock/tock/pull/1636), so importing this patch may be useful for your debugging.