google/OpenSK #69: It seems that something went wrong. App/example not found on your board.

stephenhibbert, 2020-03-06:
The OS flashed successfully but when installing the application onto the dongle I get the error below:

dev@ubuntu:/data/OpenSK$ ./deploy.py app --opensk
info: Updating rust toolchain to nightly-2020-02-03
info: syncing channel updates for 'nightly-2020-02-03-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Erasing all installed applications
Finished in 0.878 seconds
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.11s
info: Flashing padding application
Finished in 0.518 seconds
info: Installing Tock application ctap2
Finished in 1.328 seconds
Finished in 0.916 seconds
error: It seems that something went wrong. App/example not found on your board.

jmichelp, 2020-03-06:
This usually means that there's a connection issue between the JLink programmer and the board itself.

stephenhibbert, 2020-03-06:
Thankyou! Turns out I wasn't pushing down hard enough on the programmer :)

**info: You're all set!**

closed 2020-03-06 by stephenhibbert