google/OpenSK #76: Error when deploying to nrf52840_dongle

RubensZimbres, 2020-03-10:
I will use Nordic nrf52840 dongle for 2FA + security key on Google login. I successfully run setup.sh, installed rustup, tockloader (via github) but when i run:

```./deploy.py os --board=nrf52840_dk```

I get the following error:

       Finished release [optimized + debuginfo] target(s) in 0.25s
[STATUS] Flashing binaries to board
[ INFO  ] Using known arch and jtag-device for known board nrf52dk
[ERROR ] ERROR: Cannot find JLink hardware. Is USB attached ?
make: *** [flash] Error 1
fatal: Failed to execute make: Command '['make', '-C', 'third_party/tock/boards/nordic/nrf52840_dongle', 'flash']' returned non-zero exit status 2.

I tried to update the dongle driver with JLink driver but it's not working. Windows recognizes it and it is currently on port COM3

When I open JLink.exe, it tries to connect via USD but a pop up opens; "No emulators connected via USB" and it wants to connect via TCP/IP.

What are the possible solutions to this issue ? Nordic nrf52840_dongle is connected to USB port.

fateyan, 2020-03-10:
Hi, do you have a Segger J-Link connect to your computer?

You could check [detailed installation guide](https://github.com/google/OpenSK/blob/master/docs/install.md), some hardware debugging tools shows in the prerequisite.

If you want to deploy OpenSK without debugging tools, you should check this thread #3

jmichelp, 2020-10-12:
Closing the bug as many things have changes since.
Feel free to reopen if needed

closed 2020-10-12 by jmichelp