google/OpenSK #79: App object has no attribute get_binary

mweinelt, 2020-03-12:
## Expected Behavior

`deploy.py` should build the image to be flashed.

## Actual Behavior

It fails with an AttributeError, because there is no function defintion for `get_binary()`.

```
% ./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none
info: Updating rust toolchain to nightly-2020-02-03
info: syncing channel updates for 'nightly-2020-02-03-x86_64-unknown-linux-gnu'
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_mdk_dfu
    Finished release [optimized + debuginfo] target(s) in 0.49s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.64s
info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 772, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 645, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 595, in run
    self.create_hex_file(dest_file)
  File "./deploy.py", line 508, in create_hex_file
    app_tab.extract_app(board_props.arch).get_binary(),
AttributeError: 'App' object has no attribute 'get_binary'
```

## Steps to Reproduce the Problem

1. ./setup.sh
2. ./deploy.py --board=nrf52540_mdk_dfu --opensk --programmer=none

## Specifications

- Version: ec1ce661d3dde69ea2139d32def21a2fdf9c9362
- Platform: Linux x86_64

jmichelp, 2020-03-12:
This is due to tockloader project. The last version available on PyPI is v1.3 and we need v1.4 which isn't released yet.
Brad mentioned he's working on releasing v1.4

diegobes, 2020-03-16:
As a workaround:
1- uninstall tockloader: pip3 uninstall tockloader
2- clone tockloader repository
3- from the tockloader repository root: `python3 setup.py install`
This will install the latest version from master with App.get_binary support

PR #100 (2020-04-28): Depends on tockloader 1.4.x from now
Fixes #79 

tockloader 1.4.0 has been released and pushed to pypi.
Now we require at least tockloader 1.4.x for our `deploy.py` script.
Github workflows have been updated too.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-04-28 by gendx