google/OpenSK #81: Requests for USB dongle  without additional hardware.

xaqfan, 2020-03-13:
Very excited about yesterday's update。
I very much expect to get rid of the hardware (jlink) dependency。
I have one usb dongle (52840 dongle)，and have a try with nrfutil.

./deploy.py --board=nrf52840_dongle --opensk --programmer=nordicdfu
fatal: This board doesn't support flashing over DFU.

then I checked deploy.py file.Then changed parameters:
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu

I see ,it needs nrfutil command with version 5.1.
AttributeError: 'dict' object has no attribute 'iteritems'。

Can you tell me if my operation is correct，and nrfutil's version.

Thanks !

## Specifications

- Version:
- Platform: ubuntu 16.04 i686 ,python 3.5.

jmichelp, 2020-03-13:
It seems that your version of `nrfutil` is too old and doesn't support python3 (`iteritems` is a python2 thing that got removed in python3).
Try to install a newer version by running the following command:
```pip3 install --user --upgrade nrfutil```

I haven't been actively documenting the DFU flashing yet because it comes with some quirks/limitations: 
* when I tried installing `nrfutil` on my computer, it wasn't working because Nordic produced a broken wheel package for `pc_ble_driver_py` which is a dependency for `nrfutil`. Installing it on Linux required a bit of fiddling.
* the nrf52840 dongle may not be able to return to DFU mode once flashed (unlike the MDK dongle). I still have to check which pin Nordic used in their bootloader.
* I haven't tested it on other platforms than Linux at the moment.
* it requires to use Tockloader v0.4 which is not released yet. And once released, it's likely that some API/parameters have changed and that I need to adapt again the `deploy.py` script.

diegobes, 2020-03-13:
Hi;
I was trying to day with same and additional errors.
I also hit the `./deploy.py --board=nrf52840_dongle --opensk --programmer=nordicdfu
fatal: This board doesn't support flashing over DFU.` problem, and this is an easy fix in the docs ;).
The other problem is the python version needed. It looks like nrfutil will not install with python3. Installing it with python2 will bring other issues.

I was testing on OSX.
I will try with J-LINK EDU when it arrives

xaqfan, 2020-03-14:
Hi 
I'm very excited to see your reply, thanks!
I have installed  pc_ble_driver_py.　It really It took me a long time, energy。

but ,there is a another issue. could you tell me how to reslove it ?

./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
info: Updating rust toolchain to nightly-2020-02-03
info: syncing channel updates for 'nightly-2020-02-03-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle_dfu
    Finished release [optimized + debuginfo] target(s) in 0.01s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.03s
info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 776, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 649, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 599, in run
    self.create_hex_file(dest_file)
  File "./deploy.py", line 508, in create_hex_file
    app_tab.extract_app(board_props.arch).get_binary(),
AttributeError: 'App' object has no attribute 'get_binary'

It reports that  function 'get_binary' is not in 'App'

This  python's version  is  3.6.9.
Another pc's  python version is  3.5.1，don't report it!

I find many methods.

xaqfan, 2020-03-14:
> * it requires to use Tockloader v0.4 which is not released yet

You said ,you use 0.4 version..Where i can find this version.
My tockloader is tockloader-master in github.

xaqfan, 2020-03-14:
Could you flash tock os and app in your nordic 52840 dongle without hardware successfully?
Can it support flash without hardware currently?

diegobes, 2020-03-14:
I was able to compile and flash using the DFU method in my Mac.
My settings
1. python 3.7 or newer
2. install nrfutils using pip3
3. uninstall tockloder installed by OpenSK ( pip3 uninstall tockloader )
4. clone the master repository of tockloader and run setup.py install with python3. This will fix the App.get_binary issue
5. reinstall libusb: brew rm libusb --force  & brew install libusb
6. and try again:  `./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu`

Note: The dongle will be flashed and if you dont remove it form the USB you can run ioreg -p IOUSB and look for :
`    +-o OpenSK@14200000  <class AppleUSBDevice, id 0x1000277d6, registered, matched, active, busy 0 (0 ms), retain 12>`
If you instead see:
`+-o Open DFU Bootloader@14200000  <class AppleUSBDevice, id 0x10002776e, registered, matched, active, busy 0 (1 ms), ` 
it means the flash did not work.

In my case, if I remove and re insert the dongle, it resets to DFU mode and I have to flash it again.
It looks like a known issue with some 'size' parameter during flashing.

Let me know if you have any question.

xaqfan, 2020-03-15:
First thanks!
My case  is the same as your!
I can flash,and check it with ioreg -p IOUSB .
I want to know when i can flash  and use it。

Confined-, 2020-03-15:
According to [this comment](https://github.com/google/OpenSK/issues/3#issuecomment-586510300) flashing Tock a second time is needed to prevent the dongle from entering DFU mode after reinsertion.

Confined-, 2020-03-15:
I just flashed with `./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --no-app --dont-clear-apps` after the initial flash. The key is now able to be removed and reinserted without being stuck in bootloader mode.

jmichelp, 2020-03-15:
Interesting.
I will make some tests on my side and try to read the source code of Nordic bootloader to confirm this. And I'll patch the `deploy.py` script accordingly.

diegobes, 2020-03-15:
Thanks @Confined- 
The workaround sequence will be:
```
./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu
./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --no-app --dont-clear-apps
```

xaqfan, 2020-03-16:
Thanks every one .
I can flash it and work normally!

closed 2020-03-16 by xaqfan

ejeyaseelan, 2020-04-18:
so on ubuntu 18.04, is there a possibility to program the nrf52840 dongle over just usb wo needing special hardware or is that still unachievable?

anyone had any luck on non-mac so far?

matt1tk, 2020-04-19:
Looking for same thing @ejeyaseelan. Seems like it is possible, don't see why it would be only Mac. Thinking of purchasing one soon to test.

securewebsite, 2020-04-19:
> Thanks @Confined-
> The workaround sequence will be:
> 
> ```
> ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
> <<<< Re insert the Dongle >>>>
> ./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --no-app --dont-clear-apps
> ```
> 
> I can confirm this works !!!

This worked for me too. Thanks :)

matt1tk, 2020-04-19:
@securewebsite Mac or linux? and what distro if on linux?

jmichelp, 2020-04-19:
At the moment, we don't recommend to use DFU to flash a Nordic nRF52840 dongle because we haven't investigated how to go back to DFU mode once flashed.

If you don't want to use an external programmer, I suggest you pick a Makerdairy nRF52840 dongle. We added support for it too and this one can be forced to DFU mode by holding the button while plugging the dongle, allowing to flash newer versions of the firmware

Confined-, 2020-04-19:
The Nordic dongle has a reset switch near the main button, it is quite small and easy to miss. Pressing it while the dongle is plugged in will activate DFU mode. Definitely less convenient than the Makerdiary dongle but still works well enough.
[image]

jmichelp, 2020-04-19:
Ok, that's weird because this didn't work for me when I tried.
I'll give it a second try and if this works, then I'll finish the proper fix for DFU flashing

ejeyaseelan, 2020-04-20:
@jmichelp looking forward to that! 

if the softdevice from nordicsemi is incorporated into the nrf52840 dongle os + app package, then I am guessing pressing the button wont even be needed, most likely.

> This worked for me too. Thanks :)

which os?

ejeyaseelan, 2020-04-22:
So, I got somewhere with this...

On Ubuntu 18.04 and using a particle debugger, I was able to program the nrf52840 dongle using the pyocd flash option in the deploy tool. 

Now, i can login to a website simply by pressing the white button on thd dongle. 

This works!!

Particle usb debugger $12usd, on sale. 
https://store.particle.io/products/particle-debugger

mikejrh, 2020-04-28:
can anyone confirm if one can use DFU without external hardware to flash Nordic nRF52840 dongle ?

jmichelp, 2020-04-28:
You can do it with the workaround that has been mentioned (flashing everything, then flashing again the kernel only).
But you may not be able to flash an upgrade or go back to DFU until we fixed a few issues.

We're working on improving the situation.

matt1tk, 2020-04-30:
I have been unable to replicate the solution mentioned on Ubuntu. For now I'll just wait till it gets added to the deploy.py script natively 👍

diegobes, 2020-05-08:
> @jmichelp looking forward to that!
>
> if the softdevice from nordicsemi is incorporated into the nrf52840 dongle os + app package, then I am guessing pressing the button wont even be needed, most [...]

I tested it on a MacOS with Catalina

matt1tk, 2020-05-13:
I tried with the bypass and consistently got makefile errors (missing makefile.common). could be that i was running ubuntu live but I installed everything listed as necessary.

AS137430, 2020-05-20:
This is working for me but I needed to make a minor change to deploy.py as my tockloader was ver 1.5.0-dev.

```
def check_prerequisites(self):
    if not tockloader.__version__.startswith("1.5."):
      fatal(("Your version of tockloader seems incompatible: found {}, "
             "expected 1.4.x.".format(tockloader.__version__)))
```
I flashed first with
```
./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu
```
followed by reinserting and then:
```
./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --no-app --dont-clear-apps
```
and checked with
```
fido2-token -I /dev/hidraw3
proto: 0x02
major: 0x00
minor: 0x00
build: 0x00
caps: 0x05 (wink, cbor, msg)
version strings: U2F_V2, FIDO_2_0
extension strings: hmac-secret
aaguid: <mytokenguidvalue>
options: rk, up, noclientPin
maxmsgsiz: 1024
maxcredcntlst: 0
maxcredlen: 0
fwversion: 0x0
pin protocols: 1
pin retries: 6
```
I'm on Ubuntu 19.10. I also uninstalled tockloader and installed a git copy from tockloader's dev.

mikejrh, 2020-05-24:
Is it okay now to use DFU to flash a Nordic nRF52840 dongle or are there still issues ?

madmic1314, 2020-06-03:
Everything works up, including fido2-token -L shows the device.
Running fido2-token -I /dev/hidraw0   the put hangs and never returns. Removing the key gives an error. I have both the white and yellow leds flashing at the same time when I insert - not sure what this means.

bmbeverst, 2021-07-24:
This issue is still present. The work around of re flashing the Tock worked though.

Operating System: KDE neon 5.22 (Ubuntu 20.04)
Kernel Version: 5.8.0-63-generic (64-bit)