google/OpenSK #84: OpenSK is gone every re-insert of the dongle

diegobes, 2020-03-14:
## Expected Behavior
Onse I flash the dongle with OpenSK I'm expecting to stay loaded even if I re insert the dongle

## Actual BehavioI have to re-flash the dongle every time I insert it to the USB

## Steps to Reproduce the Problem
I install the new version of tockloader ( tockloader==1.4.0.dev0 )
1.  `./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu`
I get the finale message:
```
info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
```

Running ioreg -p IOUSB I get:
```
+-o Root  <class IORegistryEntry, id 0x100000100, retain 17>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100000327, registered, matched, active, busy 0$
    +-o BRCM20702 Hub@14300000  <class AppleUSBDevice, id 0x10002438a, registered, matched, active, busy 0 (0 ms), retain 12>
    | +-o Bluetooth USB Host Controller@14330000  <class AppleUSBDevice, id 0x1000243dd, registered, matched, active, busy 0 (0 ms$
    +-o Apple Internal Keyboard / Trackpad@14400000  <class AppleUSBDevice, id 0x10002439b, registered, matched, active, busy 0 (1$
    +-o OpenSK@14200000  <class AppleUSBDevice, id 0x1000277d6, registered, matched, active, busy 0 (0 ms), retain 12>
```
And I can test with the test web page: https://webauthn.io/
All work as expected.

But If I remove the usb dongle and re insert it It gets back to the DFU Bootloader:
Running ioreg -p IOUSB I get:
```
diegob@Diegos-MacBook-Pro OpenSK % ioreg -p IOUSB                 <<<=== BEFORE RE INSERT                                          
+-o Root  <class IORegistryEntry, id 0x100000100, retain 17>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100000327, registered, matched, active, busy 0$
    +-o BRCM20702 Hub@14300000  <class AppleUSBDevice, id 0x10002438a, registered, matched, active, busy 0 (0 ms), retain 12>
    | +-o Bluetooth USB Host Controller@14330000  <class AppleUSBDevice, id 0x1000243dd, registered, matched, active, busy 0 (0 ms$
    +-o Apple Internal Keyboard / Trackpad@14400000  <class AppleUSBDevice, id 0x10002439b, registered, matched, active, busy 0 (1$
    +-o OpenSK@14200000  <class AppleUSBDevice, id 0x1000277d6, registered, matched, active, busy 0 (0 ms), retain 12>

<<<<< Re Insert USB Dongle >>>>>>>>>>

diegob@Diegos-MacBook-Pro OpenSK % ioreg -p IOUSB 
+-o Root  <class IORegistryEntry, id 0x100000100, retain 17>
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100000327, registered, matched, active, busy 0$
    +-o BRCM20702 Hub@14300000  <class AppleUSBDevice, id 0x10002438a, registered, matched, active, busy 0 (0 ms), retain 12>
    | +-o Bluetooth USB Host Controller@14330000  <class AppleUSBDevice, id 0x1000243dd, registered, matched, active, busy 0 (0 ms$
    +-o Apple Internal Keyboard / Trackpad@14400000  <class AppleUSBDevice, id 0x10002439b, registered, matched, active, busy 0 (1$
    +-o Open DFU Bootloader@14200000  <class AppleUSBDevice, id 0x1000277f0, registered, matched, active, busy 0 (2 ms), retain 14$
```
## Specifications
Device: nrf52840_dongle
- Version:
- Platform: MacOS

securewebsite, 2020-03-28:
I am facing same issue. Please guide me how to fix the issue.

rkyymmt, 2020-04-07:
I am facing same issue.

xaqfan, 2020-04-27:
Have you solved this problem？ I  have checked nordic office doc some days,and no ideas.

LeZerb, 2020-04-27:
Same issue here.

LeZerb, 2020-04-27:
I suspect there might be some secure boot issue. I tried adding --sd-boot-validation=NO_VALIDATION and --app-boot-validation=NO_VALIDATION to the nrfutil call - which did not resolve the issue.
Also after having a look at the memory map for nRF52840 at https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_nrf5_v16.0.0%2Flib_secure_boot.html&cp=7_1_3_5_0_8 - I changed layout.ld so that both rom and prog were inside the application area (0x0002 7000 - 0x000F 8000 (836 kB)). After that change I was still able to flash but this the Dongle did not show up at all as a USB device. To reenter DFU mode I now need to push the reset button. So it seems that DFU mode is left but the Application is not started correctly.

Now I'm a bit lost. I suspect that the softdevice "partition" is now corrupt. Any suggestions as to what I could try next? I guess I need to download the SDK and reflash a softdevice firmware.

One open question remains on my side. Is there a way to know which bootloader version is currently flashed? I would like to try to generate bootloader settings which disable boot validation - but for that I need to know what parameter to give for --bl-settings-version.

siroccal, 2020-05-04:
Workaround https://github.com/google/OpenSK/issues/81#issuecomment-599177192 and https://github.com/google/OpenSK/issues/81#issuecomment-599263994.

geofli, 2020-06-16:
This bug should have been solved after June 10. 

`./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu`

is enough now.

geofli, 2020-06-16:
#115

closed 2020-06-16 by jmichelp