google/OpenSK #86: OpenSK source  recover without reset button.

xaqfan, 2020-03-16:
How can I  recover  dfu mode and reflash  without reset button.
My board is nordic52840 dongle (PCA10059).

I have checked office doc,and find dfu trigger library.
But i not sure the library's name,and source is contains it or not.

- Version:
- Platform:

jmichelp, 2020-03-16:
I already mentioned this here, that's the 2nd bullet point: https://github.com/google/OpenSK/issues/81#issuecomment-598654588

On the Makerdiary, if I'm correct, maintaining the button pressed while inserting the dongle does the job. I tried many things with the Nordic dongle without any luck so far. I have to look on the source code which button/GPIO they assigned to this feature.

tabroughton, 2020-03-19:
> I have to look on the source code which button/GPIO they assigned to this feature.

I'm in the same position, I flashed my dongle and didn't succeed but I lost DFU mode.  I've currently got micropython on there which I've flashed with SWD (nucleo).  But I'd like to get my DFU mode back so I can try the new deploy.py you're working on.

Looking at schematic I see RESET is attached to GPIO 0.18, 0.19, 0.21, 0.23 and 0.25 so I'm guessing pulling any of those LOW would trigger DFU mode if using [the nordic DFU trigger library](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Flib_dfu_trigger_usb.html)

xaqfan, 2020-04-10:
But , how to use nordic dfu trigger library in opensk source code ?

githubquestions, 2020-04-21:
hello basic question - how to know if lost DFU mode in failed flash? Using `/deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu` from @xaqfan fork. Was getting `Couldn't find any DFU device on your system.` Unsure if flash failed and DFU mode lost or if never worked

githubquestions, 2020-04-22:
hello basic question - how to know if lost DFU mode in failed flash? `Using /deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu` from @xaqfan fork. Was getting Couldn't find any DFU device on your system. Unsure if DFU mode lost or if never worked

update: turns out just needed libusb reinstall. Going to delete because likely off-topic

xaqfan, 2020-04-27:
> I'm in the same position, I flashed my dongle and didn't succeed but I lost DFU mode. I've currently got micropython on there which I've flashed with SWD (nucleo). But I'd like to get my DFU mode [...]

Have you solved this problem？ I have checked nordic office doc some days,and no ideas

BigPictures, 2020-05-15:
This is a feature of the [Makerdiary / Adafruit bootloader](https://github.com/makerdiary/uf2-bootloader). It claims to work on the Nordic boards as well.

> On the Nordic PCA10059 Dongle board, DFU is connected to the white button. FRST is connected to pin 1.10. Ground it to pull FRST low, as if you had pushed an FRST button. There is an adjacent ground pad.

closed 2021-11-19 by kaczmarczyck