google/OpenSK #88: NRF5840 Dongle only works when Jlink is connected

superderon, 2020-03-20:
## Expected Behavior
NRF5840 Dongle works with/without Jlink is connected.

## Actual Behavior
NRF5840 Dongle only works when Jlink is connected. This is the first time I tried to compile in dongle. Am i missing something here?

## Steps to Reproduce the Problem

1. Clone latest master repo
2. Run `./setup.sh`
3. Run `./deploy.py --board=nrf52840_dongle --opensk --programmer=jlink`

## Specifications

- Version: Latest commit: https://github.com/google/OpenSK/commit/284b1426ecaf71d7a9b2e1b445ed55953efbd370
- Platform: Linux Elementary OS, Jlink Pro

jmichelp, 2020-03-20:
I'm not sure I fully understand what's happening.

If you set the programmer to be JLink, then yes, the script will indeed expect a JLink programer to be hooked to the dongle in order to flash it. But once flashed you don't need the JLink programer to use the dongle (unless if you want to flash it again).

superderon, 2020-03-20:
> I'm not sure I fully understand what's happening.
>
> If you set the programmer to be JLink, then yes, the script will indeed expect a JLink programer to be hooked to the dongle in order to flash it. [...]

Sorry maybe the title is misleading here.
After flashing, I managed to be able to use OpenSK with my dongle, the led flashing when the web request for authentication. Pressing reset blinks the green led. (This is happens with Jlink still connected)

But then i remove the Jlink, the dongle is dead, no more led flashing when the website request for authentication. Pressing reset but nothing happens with the led.

superderon, 2020-03-20:
Seems like the dongle not detected. When i connect the Jlink, i can run `lsusb` 

and i got `Bus 001 Device 030: ID 1915:521f Nordic Semiconductor ASA `

but when i remove the jlink, its gone

ibndias, 2020-04-09:
I use another dongle and it works fine, so it's a hardware problem.

closed 2020-04-09 by ibndias

ibndias, 2020-04-10:
Just for information, I found that one of my dongle component is missing, and it's causing my dongle undetected.
[image]
Seems like I've pushed the Tag-Connect TC2050 retainer clip too hard while flashing it.