google/OpenSK #90: OpenSSH key generation fails if OpenSK has a pin set.

Confined-, 2020-03-26:
## Expected Behavior
Key generation proceeds after entering the correct pin at the prompt.

## Actual Behavior
After entering the set pin, openssh errors out with `Key enrollment failed: requested feature not supported` 

`brittle@archdesktop .ssh]$ ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk`
`Generating public/private ecdsa-sk key pair.`
`You may need to touch your authenticator to authorize key generation.`
`Enter PIN for authenticator:` 
`Key enrollment failed: requested feature not supported`

If OpenSK does not have a pin set, the key pair is generated as expected.

## Steps to Reproduce the Problem

1. Set a pin on OpenSK. I used [https://demo.yubico.com/playground](url) with passwordless login on Chromium to set the pin. 
2. Try to create a fido backed key pair with OpenSSH. 
`ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk`

## Specifications

- Version: 284b1426ecaf71d7a9b2e1b445ed55953efbd370
- Platform: Arch Linux

Confined-, 2020-03-26:
A more verbose log.

> [brittle@archdesktop ~]$ ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk -v
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x01, challenge len 0
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device /dev/hidraw0
debug1: ssh_sk_enroll: fido_dev_make_cred: FIDO_ERR_PIN_REQUIRED
debug1: sshsk_enroll: provider "internal" returned failure -3
debug1: ssh-sk-helper: Enrollment failed: incorrect passphrase supplied to decrypt private key
debug1: ssh-sk-helper: reply len 8
debug1: client_converse: helper returned error -43
Enter PIN for authenticator: 
debug1: start_helper: starting /usr/lib/ssh/ssh-sk-helper 
debug1: sshsk_enroll: provider "internal", device "(null)", application "ssh:", userid "(null)", flags 0x01, challenge len 0 with-pin
debug1: sshsk_enroll: using random challenge
debug1: ssh_sk_enroll: using device /dev/hidraw0
debug1: ssh_sk_enroll: fido_dev_make_cred: FIDO_ERR_UNSUPPORTED_ALGORITHM
debug1: sshsk_enroll: provider "internal" returned failure -2
debug1: ssh-sk-helper: Enrollment failed: requested feature not supported
debug1: ssh-sk-helper: reply len 8
debug1: client_converse: helper returned error -59
Key enrollment failed: requested feature not supported

kaczmarczyck, 2020-04-08:
Thank you for reporting, I was able to replicate the issue. The culprit is
https://github.com/google/OpenSK/blob/master/src/ctap/data_formats.rs#L501 .

If you comment that line out, everything is working as intended. I'll look into OpenSSH to check why they report a different algorithm in that case. Until then, commenting out lines 499-502 should help.

kaczmarczyck, 2020-04-08:
OpenSSH sends a COSE_ES256 == -7 defined here:
https://github.com/Yubico/libfido2/blob/780ad3c258aea5028b7b94c6623a96da3fd55224/src/fido/param.h#L72
We expect COSE_ECDH_ES256 == -25, since the specification has an exception here:

https://fidoalliance.org/specs/fido2/fido-client-to-authenticator-protocol-v2.1-rd-20191217.html#authenticatorClientPIN
"Note: The COSEAlgorithmIdentifier used is -25 (ECDH-ES + HKDF-256) although this is NOT the algorithm actually used. Setting this to a different value may result in compatibility issues."

I see different ways to resolve this:
1) Accept both the true and the fake algorithm identifier in OpenSK.
2) "Fix" OpenSSH to use the wrong CTAP2 identifier.
3) Check with FIDO how badly things break if the exception is removed.

@jmichelp Thoughts?

jmichelp, 2020-04-08:
I would go for (1) because it will take time to do (2) and that new packages are produced and deployed everywhere. And in parallel, issue a PR for (2).

closed 2020-04-09 by kaczmarczyck

martelletto, 2020-04-10:
Corresponding libfido2 PR: https://github.com/Yubico/libfido2/pull/155.

PR #92 (2020-04-09): accept the algorithm identifier used by OpenSSH
Fixes #90 

There are now 2 algorithm identifiers allowed in OpenSK: The backwards compatible -25 from CTAP2, and the identifier required by OpenSSH.

This command now works as expected:
ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk