google/OpenSK #98: yapf: "chromium" is not a valid style or file path

gendx, 2020-04-24:
## Expected Behavior

`pylint/yapf` isn't affected by a pull request which doesn't modify any Python file.

## Actual Behavior

This fails on #97.

```
Run echo ::add-matcher::./.github/python_matcher.json
  echo ::add-matcher::./.github/python_matcher.json
  yapf --style=chromium --recursive --exclude third_party --diff .
  shell: /bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.7.6/x64
yapf: "chromium" is not a valid style or file path
##[error]Process completed with exit code 1.
```

## Steps to Reproduce the Problem

See https://github.com/google/OpenSK/pull/97/checks?check_run_id=614885048

jmichelp, 2020-04-24:
Seems related to this commit: https://github.com/google/yapf/commit/22ef70f3c497436adf59934b2ffa84ce6f962b88

I have to do some test then. Maybe we'll move to the "google" style then.

jmichelp, 2020-04-24:
After some test, the provided google style moves to 4 spaces soft tabs. So I picked the yapf style which is now the replacement for the chromium one.

PR #99 (2020-04-24): Fix style being used with yapf 0.30.0
Fixes #98 

Chromium team moved to PEP8 and deleted their custom style in yapf, which released a 0.30.0 version on pypi.
OpenSK follows by moving to yapf default style which is now the one used by Chromium too.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-04-24 by jmichelp