google/OpenSK #106: Implement CTAP 2.1

kaczmarczyck, 2020-05-06:
The review draft for CTAP 2.1 adds new commands and requires seom changes in the old ones.
https://fidoalliance.org/specs/fido2/fido-client-to-authenticator-protocol-v2.1-rd-20191217.html
We want to support new features like deleting resident keys.

Our access to a more recent version of CTAP 2.1 allows us to prototype those new features and give feedback on the command design. This issue tracks our progress. Please be aware that these commands are still exprimental and the implementation can change in the future. Do not rely on them!

kaczmarczyck, 2021-01-27:
The current implementation in the `develop` branch follows [review draft 2 of CTAP 2.1](https://fidoalliance.org/specs/fido-v2.1-rd-20201208/fido-client-to-authenticator-protocol-v2.1-rd-20201208.html).

PRs merged for finishing support:

- #254
- #256 
- #263 
- #265 
- #267 
- #273 

Current WIP:
#274 

Missing work:

- updating all existing command algorithms
- `alwaysUv` for `MakeCredential` and `GetAssertion`
- `toggleAlwaysUv` in `authenticatorConfig`
- enterprise attestation
- `enableEnterpriseAttestation` in `authenticatorConfig`
- PIN Protocol V2
- all potential changes after review draft 2

kaczmarczyck, 2021-03-22:
Features around `alwaysUv`, enterprise attestation and PIN protocol V2 (PRs for the latter: #293, #295, #296) are implemented. Missing work:
- remaining updates for existing command algorithms
- missing changes from [review draft 3](https://fidoalliance.org/specs/fido-v2.1-rd-20210309/fido-client-to-authenticator-protocol-v2.1-rd-20210309.html)

kaczmarczyck, 2021-04-06:
CTAP2.1 changes are complete with #297 and #299 (to be reviewed as of now). Not all features are thoroughly tested yet.

Sami32, 2022-08-27:
Any update on this?

It seem that features have been tested and that CTAP 2.1 get its final version:
https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-errata-20220621.html

kaczmarczyck, 2022-08-29:
Hi! We haven't noticed any problems with our implementation of CTAP 2.1 yet. It will take more time for an official release, but you can try out the `develop` branch if you are curious about the current state. Is there anything specific you are interested in?

Sami32, 2022-09-10:
Hi!
Sorry for my late answer.
Thanks for the information +1
I'm mostly interested in the management of discoverable credentials.

Note:
https://eprint.iacr.org/2022/1029

closed 2023-06-12 by kaczmarczyck