google/OpenSK #115: OpenSK firmware can not work well with nRF52840 dongle after 2e419fe77b37255778fe13166cdc891caee3b334

Geoffrey-Li, 2020-06-09:
## Expected Behavior
Before (include) 2e419fe77b37255778fe13166cdc891caee3b334, when insert nRF52840 USB dongle to USB port, LEDs are off. When receiving CTAP2/U2F command, LEDs will blink and wait for user presence. After user press the button, it response correctly.

## Actual Behavior
Now (after 2e419fe77b37255778fe13166cdc891caee3b334), after insert to USB port, LEDs are blinking even if it is in idle. OpenSK does not work after user presses button.

## Steps to Reproduce the Problem

1. Download firmware to nRF52840 dongle as usual.
2. Replug it to USB port, take a look at the LED status, they are blinking, different from before.
3. Test it on https://webauthn.io/ or other test website.
4. Make credential and press button.

I tested source code from 752bf47ed5caa2265e1d3884a40798fd7a2e95d6 to latest. It has same behavior.

kaczmarczyck, 2020-06-09:
Thank you for your report! The blinking might be a kernel panic? Is the blinking pattern different from before?
I'll assign to @ia0 because of the commit number for now.

ia0, 2020-06-09:
The commit number is surprising, but there's indeed high chance that this is due to recent changes on the storage (in particular the fact that it moved outside and after the app).

@Geoffrey-Li When syncing at current HEAD (642ee69), one would need to do the following things for a clean install:
1. `./reset.sh &&./setup.sh` before compiling OpenSK (but I guess you already did so otherwise it should not compile).
2. Some way to erase the pages after 0xC0000. There is a `--clear-storage` flag in the `deploy.py` script but the dongle is not supported yet. One solution I would see for now is to append `0xff` padding at the end of the initial HEX file until at least 20 pages after address 0xC0000 (i.e. until 0xE4000 but 0x100000 would be even safer).

Geoffrey-Li, 2020-06-09:
@kaczmarczyck , yes, using the source code after 2e419fe77b37255778fe13166cdc891caee3b334, in idle status, the LEDs blink quickly just like it receives CTAP2/U2F command.

@ia0, I download firmware though dfu. I'll try to clear all flash by using J-LINK before dfu.

Geoffrey-Li, 2020-06-09:
@ia0 , I used J-LINK to clear all flash, 1024K+separate 4K, and then downloaded Open bootloader, downloaded OpenSK through DFU. Same behavior. LEDs blinks and CTAP2/U2F can not work.

ia0, 2020-06-09:
Then this looks unexpected. I can't reproduce on a dev-kit. I'll try to see if I can reproduce on a dongle tomorrow.

innofocus, 2020-06-09:
Same issue for a genuine Nordic usb dongle : leds 1 and blue are blinking simultaneously around 200ms and no success on testing at webauthn.io.
I've been trying with half success commit 2e419fe and branch HMAC-secret-extension : 
- if dongle not unplugged once, it works - tested on webauthn.io.
- if dongle unplugged once, replug puts it in DFU mode - red slowly blinking.

Therefore, there are 2 issues for me : 
- 2 leds blinks and not working at all - this issue
- And : erroneous flash - it looks like it's been used --no-persistent-storage option at deploy.sh

innofocus, 2020-06-09:
Odd issue : when passing "--no-persistent-storage", the key works well (commit 2e419fe) without default to DFU behavior anymore !

innofocus, 2020-06-09:
One more action : master branch and --no-persistent-storage seems to solve the issue...
git checkout master; ./reset.sh; ./setup.sh
and : 
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu --no-persistent-storage

The dongle work well at webauthn.io, even when unplugged before !

ia0, 2020-06-09:
Thanks for the investigation. The fact that `--no-persistent-storage` fixes the issue is a good sign that this has to do with the fact that the persistent storage moved out of the application (the storage probably fails to initialize for some reason). What is still unclear to me is why this only affects the dongle (or the DFU way of flashing the dongle). I'll look into this tomorrow.

ia0, 2020-06-10:
I could reproduce the issue on a dongle. But it is still unclear what is the cause of the issue.

I tried to reproduce on a dev-kit using a DFU and flashing a merged hex file with a modified dev-kit board to support DFU. This experiment did not show any issue. So the problem may be:
- When using the DFU to flash the hex file (since flashing the hex file directly works on dev-kit).
- Something specific to the dongle.

ia0, 2020-06-10:
The problem was indeed specific to the dongle. The board of the dongle did not define the storage location. #119 fixes the issue.

geofli, 2020-06-11:
Thank you very much. I checked with latest source code, this issue and #81 have been solved.