google/OpenSK #125: Add a CI workflow to run cargo-outdated checks

gendx, 2020-06-26:
The [cargo-outdated](https://github.com/kbknapp/cargo-outdated) tool checks for outdated dependencies in a Cargo.toml. As a complement to cargo-audit which checks for vulnerable dependencies, we should leverage that to make sure dependencies are up-to-date.

jmichelp, 2020-08-20:
Without the `--depth` argument, it's too verbose.

But this seems to be fine:
```cargo outdated --depth 1
Name     Project  Compat  Latest  Kind         Platform
----     -------  ------  ------  ----         --------
elf2tab  0.4.0    ---     0.6.0   Development  ---
```

jmichelp, 2020-10-14:
Blocked by https://github.com/kbknapp/cargo-outdated/issues/241

Not only the produced JSON output is truncated (resulting in an empty JSON when it shouldn't be for example) but the produced JSON is also changing from a run to another.

kaczmarczyck, 2023-11-09:
Now that we pin `Cargo.lock` files, we don't want outdated notifications. Instead, we will upgrade compiler versions and packages before a release (or earlier, if cargo audit notifies us of problems).

Will be fixed with #663 .

closed 2023-11-09 by kaczmarczyck