google/OpenSK #126: Test UTF-8 encoding of newPin and curPin

geofli, 2020-07-02:
According to CTAP 2.0 Proposed Standard: https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html

and 

CTAP 2.1 Review Draft: https://fidoalliance.org/specs/fido2/fido-client-to-authenticator-protocol-v2.1-rd-20191217.html

and latest working draft, to authenticatorClientPIN:
 "newPin" be the UTF-8 representation of "newPinUnicode".
 "curPin" be the UTF-8 representation of "curPinUnicode" 

So authenticator should check newPin and curPin against UTF-8 encoding, then test tool should test this with wrong/correct UTF-8 encoding input parameters.

kaczmarczyck, 2020-07-02:
Thank you for the report! I think the specification is ambiguous when it comes to UTF-8 encoding, but WebAuthn has a more pronounced opinion:
https://w3c.github.io/webauthn/#sctn-strings i.e.

> Such truncation MAY also respect UTF-8 sequence boundaries or grapheme cluster boundaries [UTR29].

CTAP2, in my opinion (and I checked with others), only requires platforms to check for encoding:

> **Platform** checks the Unicode character length of "newPinUnicode" against the minimum 4 Unicode character requirement

Your quote seems to describe platform behaviour. Enforcing correct UTF-8 in the test tool for 2.0 would not be passed by most security keys out there. In 2.1, there is some understanding of the encoding required, the new minPinLength feature is:

> The minimum PIN length in Unicode code points that authenticator enforces.

What do you think?

geofli, 2020-07-02:
Yes, you are right, platform should check that; But, in the point of view of authenticator, if it receives wrong encoded newPin or curPin anyhow, what should it do? For example, test tools or applications which can sent command to authenticator directly.

kaczmarczyck, 2020-07-02:
That depends on the type of authenticator. If you have a display, you want to check the strings for proper encoding and possibly return an error when it notices that platform's didn't properly do their job. I'd propose the error code `CTAP2_ERR_INVALID_CBOR` since correct CBOR normally encompasses UTF-8 checks. 

If your authenticator does not have a display, it treats text strings as opaque byte strings. All text strings in the specification are either store-and-forward (like user name, RP icon) or later treated as byte strings (like RP ID, PIN).

geofli, 2020-07-06:
Sounds good, although it is strange that the authenticator clearly knows the string is wrong formatted but do nothing.

kaczmarczyck, 2020-07-06:
Then I'll leave this open until correct UTF8 behavior is enforced for CTAP2.1.

kaczmarczyck, 2021-01-29:
Fixed in #246 in the function `check_and_store_new_pin` in `src/ctap/pin_protocol_v1.rs`. We now check:

```rust
    let pin_length = str::from_utf8(&pin).unwrap_or("").chars().count();
    if pin_length < min_pin_length || pin.len() == PIN_PADDED_LENGTH {
        return Err(Ctap2StatusCode::CTAP2_ERR_PIN_POLICY_VIOLATION);
    }
```

closed 2021-01-29 by kaczmarczyck