google/OpenSK #130: Running deploy.py with --programmer=none results in error; hex files not created

jasonloveday, 2020-07-23:
Hi there,

I am fairly new to OpenSK, so please be gentle.

## Expected Behavior
Running ./deploy.py with --programmer=none should create hex files for manual programming

## Actual Behavior
Compiling appears to work, error message as follows:

info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 901, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 734, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 684, in run
    self.create_hex_file(dest_file)
  File "./deploy.py", line 584, in create_hex_file
    app_tab.extract_app(board_props.arch).get_binary(),
TypeError: get_binary() missing 1 required positional argument: 'address'

## Steps to Reproduce the Problem

1. ./deploy.py --board=nrf52840_dongle --opensk --programmer=none

## Specifications

- Version: unknown
- Platform:

gendx, 2020-07-23:
We have something very similar in CI (https://github.com/google/OpenSK/blob/master/reproduce_board.sh#L27). This looks like using an non-supported version of the [Tockloader library](https://pypi.org/project/tockloader/#history).

Can you run `./reset.sh && ./setup.sh` before re-running this command?

gendx, 2020-07-23:
I checked again, and it looks like the latest version of Tockloader (1.5.0) that was just released this week broke the `deploy.py` script on our side.

jasonloveday, 2020-07-23:
Excellent, thanks!

PR #136 (2020-08-11): Move to tockloader 1.5
Fixes #130 

Just moving to a newer version of tockloader.
We don't need anymore to craft a fake app for padding as tockloader 1.5 understand what padding is and provides a dedicated object.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2020-08-11 by jmichelp