google/OpenSK #133: GnuPG/OpenSSH and KeePassXC

CarrotK, 2020-07-25:
Sorry for the stupid question!

Is it planned in the future to support offline services such as: 
1. HMAC-SHA1 (challenge-response pair) - for using in KeePassXC
2. ed25519-sk - in OpenSSH (because ecdsa-sk has support)
3. Support for LUKS on Linux (something like "yubikey-luks and yubikey-personalization" )

P.S. Excuse me for bad English :)

CarrotK, 2020-07-25:
> Sorry for the stupid question!
>
> Is it planned in the future to support offline services such as:
>
> 1. HMAC-SHA1 (challenge-response pair) - for using in KeePassXC
> 2. ed25519-sk - in OpenSSH (because [...]
Arm CryptoCell CC310 - support: **SHA-1, SHA-2 up to 256 bits (Keyed-hash message authentication code (HMAC) / Ed25519, Curve25519 / secp160k1, secp192k1, secp224k1, secp256k1 / secp160r1, secp192r1, secp224r1, secp256r1, secp384r1, secp521r1 / NIST FIPS 186-4 - P-192, P-224, P-256, P-384, P-521/RSA Up to 2048-bit key size / ChaCha20/Poly1305 Supported key size: 128 and 256 bits / AES-128bit**

> P.S. Excuse me for bad English :)

kaczmarczyck, 2020-07-28:
We currently do not intend to implement or maintain additional protocols. You are of course invited to fork or contribute!

closed 2020-10-12 by kaczmarczyck

wolfyzhang-github, 2023-03-15:
Tsinghua University has implemented a physical encryption key based on the same chip, which supports the GPG protocol(it passed my test), however, it is still very imperfect and many other features don't work properly, so maybe you can try this if you **only** need GPG.