google/OpenSK #134: Cygwin, or How do I use the key to login to my remote ssh servers from Windows?

jasonloveday, 2020-07-27:
Hi there,

Still fairly new to OpenSK, please be gentle...

I am trying to use the key to authenticate me to my remote ssh servers from my Windows 10 machine.

I've tried several combinations, taken from various guides I've found on the Internet, with very limited success. The closest I've managed to get is to run Cygwin on my Windows 10 machine and install OpenSSH (8.3).

The snag I've hit now is that Cygwin doesn't appear to find the token when I attempt to create the keys. Below is the output:

$ ssh-keygen -t ecdsa-sk -f ~/.ssh/id_ecdsa_sk
Generating public/private ecdsa-sk key pair.
You may need to touch your authenticator to authorize key generation.
Key enrollment failed: device not found
</snip>

Doing a 'lsusb' within my Cygwin environment shows that it can see the USB device:

$ lsusb
Bus 001 Device 016: ID 1915:521f Nordic Semiconductor ASA

I'd be grateful for any suggestions anyone may have.

Many thanks.

## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version:
- Platform:

closed 2020-07-27 by jasonloveday

kaczmarczyck, 2020-07-27:
Hi! Did you close the issue because you solved it on your own? If you still need help, could you please answer a few questions so I understand your problem?

1. Can you give me a detailed step-by-step description what you did with your OpenSK so far?
2. You use the work "key" a few times. Can you disambiguate by using "authenticator" or "security key" for OpenSK, and "cryptographic key" or "key material" for i.e. ECDSA public/private keys?

jasonloveday, 2020-07-27:
Hi,
Yes, I solved it on my own, it turns out that Cygwin requires admin privileges to access the FIDO2 key (token). So running it as admin solved the problem. I tried to delete my question in Github but I couldn't see how to do that.

However, I am curious to know if anyone knows a way to get PuTTY to use the FIDO2 key as an authentication method?

Cheers,

kaczmarczyck, 2020-10-12:
This question doesn't seem to be OpenSK related, but rather specific to PuTTY? If you have any more questions about OpenSK, please don't hesitate to open a new issue.