google/OpenSK #141: conflicting binary elf2tab

coelner, 2020-08-13:
## Expected Behavior
If this project needs a specific version of a binary, this binary should be handled internally

## Actual Behavior
```
./deploy.py --board=nrf52840_dongle --opensk --programmer=none
info: Updating rust toolchain to nightly-2020-06-10
info: syncing channel updates for 'nightly-2020-06-10-x86_64-unknown-linux-gnu'
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle
    Finished release [optimized + debuginfo] target(s) in 0.03s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.20s
info: Generating Tock TAB file for application/example ctap2
error: Detected unsupported elf2tab version 'elf2tab 0.6.0'. The following commands may fail. Please use 0.5.0 instead.
info: Generating all-merged HEX file: target/nrf52840_dongle_merged.hex
```

## Steps to Reproduce the Problem

1. cargo install elf2tab
2. run deploy.sh

## Specifications

- Version: 57bf623
- Platform: archlinux 

## possible workaround
```
diff --git a/deploy.py b/deploy.py
index 55fd00d..9ea3de2 100755
--- a/deploy.py
+++ b/deploy.py
@@ -404,7 +404,7 @@ class OpenSKInstaller:
     assert self.args.application
     info("Generating Tock TAB file for application/example {}".format(
         self.args.application))
-    elf2tab_ver = self.checked_command_output(["elf2tab", "--version"]).split(
+    elf2tab_ver = self.checked_command_output(["./bin/elf2tab", "--version"]).split(
         "\n", maxsplit=1)[0]
     if elf2tab_ver != "elf2tab 0.5.0":
       error(
diff --git a/setup.sh b/setup.sh
index ff77b6d..8d12e74 100755
--- a/setup.sh
+++ b/setup.sh
@@ -42,4 +42,4 @@ pip3 install --user --upgrade 'tockloader==1.5' six intelhex
 rustup target add thumbv7em-none-eabi
 
 # Install dependency to create applications.
-cargo install elf2tab --version 0.5.0
+cargo install elf2tab --version 0.5.0 --root .
diff --git a/third_party/tock b/third_party/tock
--- a/third_party/tock
+++ b/third_party/tock
@@ -1 +1 @@
-Subproject commit f1f9d717a4d9b5ae934a3b50ce6a18e6dff3e7c7
+Subproject commit f1f9d717a4d9b5ae934a3b50ce6a18e6dff3e7c7-dirty

```
[projectBinary.diff.txt](https://github.com/google/OpenSK/files/5069378/projectBinary.diff.txt)

kaczmarczyck, 2020-08-13:
Our setup script installs the correct version of elf2tab:
https://github.com/google/OpenSK/blob/57bf6235394692a95c5bfcae0970fe62800f305d/setup.sh#L45

Is there an issue when you execute `./setup.sh` first?

coelner, 2020-08-13:
That's right, you install it globally. But if I need an other version, I have to reinstall those all the time. Currently the latest version 0.6.0 works and the error message (which should maybe a warning) can be ignored.

gendx, 2020-08-17:
> That's right, you install it globally. But if I need an other version, I have to reinstall those all the time. Currently the latest version 0.6.0 works and the error message (which should maybe a warning) can be ignored.

In the past, we've had breakages when a new release of tockloader or elf2tab introduces a breaking change that affects OpenSK's deploy script (e.g. #130). That's why we pin a specific version. Of course it may well be that a new release doesn't break anything at all (such as migrating elf2tab 0.5.0 => 0.6.0), but in general it shouldn't be a warning because it could indeed break the setup/deploy scripts. We cannot predict in advance what will break or won't break in future releases of external tools.

That said, the goal is to keep these pinned versions as up-to-date as possible, and we regularly do it (e.g. #136). If you want it to be prioritized, please feel free to send a pull request.

jmichelp, 2020-09-16:
I just bumped the version to 0.6.

@coelner would you mind updating your PR. I think it's a good idea to install and run the version locally.

PR #147 (2020-09-16): Few fixes/improvements
Fixes #141  by bumping elf2tab to v0.6

This PR also:
- removes mention of "Google" as part of the generated CA/certs
- fixes logic mistake to enable CTAP2.1 features
- generates a proper GUID instead of crafting one from the serial of the certificate.

- [x] Tests pass
- [ ] Appropriate changes to README are included in PR
PR #145 (2020-09-24): Elf2tab internally install
Fixes #141 

- [ ] Tests pass
- [ ] Appropriate changes to README are included in PR
closed 2020-09-24 by jmichelp