google/OpenSK #155: Investigate why the modifications on Tock regarding the MPU makes OpenSK panic

jmichelp, 2020-09-21:
## Expected Behavior

OpenSK should not panic when using the flash storage

## Actual Behavior

When moving to a more recent kernel version, it appears that the storage causes a panic because of the changes in the MPU configuration.
We 

## Steps to Reproduce the Problem

1. Use updated kernel/patches from #150 
1. Don't apply the patch `patches/tock/05-mpu-fix.patch`
1. Compile and deploy OpenSK firmware with the flag `--clear-storage`
1. kernel panic should happen immediately after the board resets.

## Specifications

- Version: N/A
- Platform: N/A

ia0, 2020-09-21:
The issue has been introduced by https://github.com/tock/tock/pull/1873. I've [commented](https://github.com/tock/tock/pull/1873#issuecomment-696217269) to get a tracking bug. That PR introduces a new concept, `KernelMpuConfig`, which is not yet implemented on any of the architectures, but necessary for a fix at least for Cortex-M.

ia0, 2020-09-24:
This bug has been triaged and fixed upstream: https://github.com/tock/tock/pull/2120. Do we want to proactively sync the kernel again to remove the MPU patch?

jmichelp, 2020-09-24:
The Time HIL also landed, potentially fixing other issues we have. Let's wait a bit more before a new sync.

kaczmarczyck, 2021-10-13:
Fixed with the kernel upgrade.