google/OpenSK #157: Remove the top-level library

gendx, 2020-09-22:
The top-level package is both a library `src/lib.rs` and a binary `src/main.rs`. This creates quite a bit of confusion, as highlighted in #149 and #156, and may cause increased binary size due to potential code duplication.

There should really only be a thin binary at the top level, with other components split into separate libraries in `libraries/` or `third_party/`.

ia0, 2020-09-22:
To clarify the confusion and situation:
- After #156, the ctap2 crate (the one at the root of the git repository) contains both a library and a binary that share the same module hierarchy.
- The potential code duplication comes from the fact that modules are shared between the library and the binary.
- Having a library and a binary in a crate is a [common](https://doc.rust-lang.org/stable/book/ch12-03-improving-error-handling-and-modularity.html#separation-of-concerns-for-binary-projects) Rust pattern for binaries.
- As such, to address the issue it is sufficient to call the library from the binary instead of copying the modules.
- It is an orthogonal consideration of whether the ctap2 library should be in another crate than the ctap2 binary. This is probably a good thing to do if the ctap2 binary doesn't end up being a single module after the step above.

kaczmarczyck, 2023-06-12:
Should be fixed with our CTAP library.

closed 2023-06-12 by kaczmarczyck