google/OpenSK #175: Stuck at 'info: Flashing HEX file'

FilipLaurentiu, 2020-10-04:
## Expected Behavior
install TockOS and certificates

## Actual Behavior
```
....
info: Building OpenSK application
    Updating crates.io index
   Compiling libc v0.2.78
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling getrandom v0.1.15
   Compiling cfg-if v0.1.10
   Compiling pkg-config v0.3.18
   Compiling cc v1.0.60
   Compiling autocfg v1.0.1
   Compiling syn v1.0.42
   Compiling bitflags v1.2.1
   Compiling libtock_core v0.1.0 (/home/filip/Desktop/OpenSK/third_party/libtock-rs/core)
   Compiling ppv-lite86 v0.2.9
   Compiling openssl v0.10.30
   Compiling foreign-types-shared v0.1.1
   Compiling lazy_static v1.4.0
   Compiling byteorder v1.3.4
   Compiling subtle v2.3.0
   Compiling linked_list_allocator v0.8.6
   Compiling cbor v0.1.0 (/home/filip/Desktop/OpenSK/libraries/cbor)
   Compiling arrayref v0.3.6
   Compiling foreign-types v0.3.2
   Compiling quote v1.0.7
   Compiling rand_core v0.5.1
   Compiling rand_chacha v0.2.2
   Compiling rand v0.7.3
   Compiling uuid v0.8.1
   Compiling openssl-sys v0.9.58
   Compiling ctap2 v0.1.0 (/home/filip/Desktop/OpenSK)
   Compiling libtock_codegen v0.1.0 (/home/filip/Desktop/OpenSK/third_party/libtock-rs/codegen)
   Compiling libtock_drivers v0.1.0 (/home/filip/Desktop/OpenSK/third_party/libtock-drivers)
   Compiling lang_items v0.1.0 (/home/filip/Desktop/OpenSK/third_party/lang-items)
   Compiling crypto v0.1.0 (/home/filip/Desktop/OpenSK/libraries/crypto)
    Finished release [optimized] target(s) in 35.06s
info: Generating Tock TAB file for application/example ctap2
info: Generating all-merged HEX file: target/nrf52840_dongle_merged.hex
info: Flashing HEX file

```

## Steps to Reproduce the Problem

1. Plug the Nordic nRF52840 Dongle
2. Run ./deploy.py --board=nrf52840_dongle --opensk --programmer=pyocd

## Specifications

- Version: commit 079c0a4
- Platform: Ubuntu 20

FilipLaurentiu, 2020-10-04:
The script is stuck at 'info: Flashing HEX file'

FilipLaurentiu, 2020-10-04:
Number of writeable flash regions: 0
  Adding .crt0_header section. Offset: 64 (0x40). Length: 64 (0x40) bytes.
Entry point is in .text section
  Adding .text section. Offset: 128 (0x80). Length: 223328 (0x36860) bytes.
  Adding .stack section. Offset: 223456 (0x368e0). Length: 16384 (0x4000) bytes.
Searching for .rel.X sections to add.
TBF Header:
               version:        2        0x2
           header_size:       56       0x38
            total_size:   262144    0x40000
                 flags:        1        0x1

        init_fn_offset:       73       0x49
        protected_size:        8        0x8
      minimum_ram_size:   107428    0x1A3A4

     start_process_ram: 4294967295 0xFFFFFFFF
   start_process_flash:   262208    0x40040

info: Generating all-merged HEX file: target/nrf52840_dongle_merged.hex
info: Flashing HEX file
Waiting for a debug probe to be connected...

FilipLaurentiu, 2020-10-04:
Running with verbose mode. I notice that is the same output if I run the command with board connected or not

```
OpenSK$ ./deploy.py --board=nrf52840_dongle --opensk --programmer=pyocd --verbose --debug --verbose-build --panic-console

..
Number of writeable flash regions: 0
  Adding .crt0_header section. Offset: 64 (0x40). Length: 64 (0x40) bytes.
Entry point is in .text section
  Adding .text section. Offset: 128 (0x80). Length: 223328 (0x36860) bytes.
  Adding .stack section. Offset: 223456 (0x368e0). Length: 16384 (0x4000) bytes.
Searching for .rel.X sections to add.
TBF Header:
               version:        2        0x2
           header_size:       56       0x38
            total_size:   262144    0x40000
                 flags:        1        0x1

        init_fn_offset:       73       0x49
        protected_size:        8        0x8
      minimum_ram_size:   107428    0x1A3A4

     start_process_ram: 4294967295 0xFFFFFFFF
   start_process_flash:   262208    0x40040

info: Generating all-merged HEX file: target/nrf52840_dongle_merged.hex
info: Flashing HEX file
Waiting for a debug probe to be connected...
```

FilipLaurentiu, 2020-10-04:
More debug info
```
filip@filip-pc:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0029 Intel Corp. 
Bus 001 Device 008: ID 1915:521f Nordic Semiconductor ASA Open DFU Bootloader
Bus 001 Device 002: ID 13d3:5666 IMC Networks USB2.0 HD UVC WebCam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
filip@filip-pc:~$ pyocd list
No available debug probes are connected
filip@filip-pc:~$ 
```

jmichelp, 2020-10-04:
Well it seems you have no OpenOCD probe connected to your computer (and the probe should be wired to the dongle too).
What about using DFU flashing, which doesn't require any external hardware?

`$ ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu`

If you haven't already installed the corresponding flashing tool, you will also need to run `pip3 install nrfutil`

FilipLaurentiu, 2020-10-04:
> Well it seems you have no OpenOCD probe connected to your computer (and the probe should be wired to the dongle too).
> What about using DFU to flash OpenSK, which doesn't require any external hardware?
> 
> `$ ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu`
> 
> If you haven't already installed the corresponding flashing tool, you will also need to run `pip3 install nrfutil`

Now I got this
```

OpenSK$ ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
info: Updating rust toolchain to nightly-2020-06-10
info: syncing channel updates for 'nightly-2020-06-10-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle_dfu
    Finished release [optimized + debuginfo] target(s) in 0.02s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.03s
info: Generating Tock TAB file for application/example ctap2
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.

info: Flashing device using DFU...

Traceback (most recent call last):
  File "/home/filip/.local/lib/python3.8/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 198, in open
    self.serial_port = Serial(port=self.com_port,
  File "/home/filip/.local/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/filip/.local/lib/python3.8/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/filip/.local/bin/nrfutil", line 8, in <module>
    sys.exit(cli())
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/filip/.local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 1014, in usb_serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, False,
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 970, in do_serial
    dfu.dfu_send_images()
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 127, in dfu_send_images
    self._dfu_send_image(self.manifest.application)
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 88, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/filip/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 202, in open
    raise NordicSemiException("Serial port could not be opened on {0}"
pc_ble_driver_py.exceptions.NordicSemiException: Serial port could not be opened on /dev/ttyACM0. Reason: could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'
```

And I already added the udev rules 

```
filip@filip-pc:~/Desktop/OpenSK$ sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
filip@filip-pc:~/Desktop/OpenSK$ sudo udevadm control --reload
```

FilipLaurentiu, 2020-10-04:
`sudo chmod a+rw /dev/ttyACM0`
fix it

FilipLaurentiu, 2020-10-04:
Thank you very much !

closed 2020-10-04 by FilipLaurentiu