google/OpenSK #176: Enable Rustfmt

gendx, 2020-10-05:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version:
- Platform:

kaczmarczyck, 2020-10-12:
As indicated, I expected this to automatically checked. No strong opinion however, where are ASCII art examples that would need refactoring?

gendx, 2020-10-12:
> As indicated, I expected this to automatically checked. No strong opinion however, where are ASCII art examples that would need refactoring?

It'd be easier to have an exhaustive list while reviewing code, but for a start `src/ctap/ctap1.rs` has some tables (not sure if it should be best described as "ASCII art" or "markdown").

kaczmarczyck, 2020-10-12:
Just wanted to know if you have an intuition about how much we have to change. But it seems like our code would be nicer with correct length comments? But I'd say it's low priority.