google/OpenSK #179: Should clear all credentials/PIN after flashing the new firmware?

geofli, 2020-10-09:
Currently, after you flash a new firmware to OpenSK, existing PIN/credentials (resident or non-resident) are still valid, you can still use them to authenticate to online services.

Should OpenSK clear all existing Credentials(resident or non-resident), PIN and other security parameters if it have been updated to the new firmware?

jmichelp, 2020-10-09:
That's working as intended to allow bug fixes/firmware upgrade without being locked out from your accounts. If you want to clear the storage, you have to use the flag `--clear-storage` when running the `deploy.py` scripts.

geofli, 2020-10-09:
Oh.., thank you very much, I never used --clear-storage.

geofli, 2020-10-09:
```
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu --clear-storage
info: Erasing the persistent storage
fatal: Programmer nordicdfu is not supported.
```

jmichelp, 2020-10-09:
I don't think we added support for it indeed. It should be technically feasible as we first clear the storage before flashing the application/kernel so this shouldn't mess up with the DFU checksum.
Until then, another way to clear the storage is to reset the key through CTAP2 protocol. But we don't provide such tool as part of the OpenSK repository.

geofli, 2020-12-22:
I highly recommend add this feature or provide a method to clear old app data during or before/after flashing the new version. I have met issue.
I have two dongles, which have flashed old version several months before. I don't remember what FIDO operations I took, maybe registered to online services or tested on WebAuthN test website. 
But today, after I flash latest OpenSK, the LEDs of one dongle wink regularly while idle. FIDO functions can not work well even though it can be recognized by OS. If I clear all storage by using J-Link and then flash OpenSK, it can work.
The other one seems OK when operate on WebAuthN test, but if I try to reset it through Chrome security key manager or Windows Sign-in options, it can not be reset.

kaczmarczyck, 2020-12-22:
Our first storage implementation wasn't update friendly, and indeed resetting OpenSK through CTAP2 is not an option when you panic because of internal update problems. We will soon have an OpenSK v1.0 that should update gracefully.
I'll assign this to @jmichelp for all open DFU related questions.

jmichelp, 2020-12-22:
Clearing the storage over DFU is a bit problematic. It involves flashing a fake slice of flash that the bootloader will think is valid code.
This could be added through a vendor specific command, although I would argue that a regular life cycle would be:
- flash an initial firmware over DFU (storage will be initialized at first boot)
- use the key
- update the firmware (storage shouldn't have to be cleared)
- use FIDO2 reset command to clear storage to erase existing credentials
- should a firmware update need to fully clear the storage this must be done by the firmware upgrade component (yet to be designed). But this would break the key if it's configured with U2F support of batch attestation activated as the certificate and private key are also part of the storage.

The situation that has been experienced was due to a transient phase of the storage format, as explained in the previous comment.
For a "production" unit, DFU will be used to flash the initial firmware but should be de-activated after that. Updating the firmware will be done without DFU or with firmware signature activated, which will prevent anyway the storage to be cleared over DFU.

ia0, 2020-12-22:
Note that as a temporary workaround we could write an example application that simply erases the storage. Then if ones wants to use the `--clear-storage` flag of `deploy.py` over DFU, they could first `deploy.py --erase_storage` then `deploy.py --opensk`. The first command installs the example application call `erase_storage` which simply erases all the storage exposed by the board (i.e. 0xC0000 to 0xFFFFF on nRF52). I can do that if needed.

We could maybe even drop the `--clear-storage` flag from `deploy.py` and only use the `erase_storage` application.

geofli, 2020-12-22:
>  a temporary workaround we could write an example application that simply erases the storage.
That is indeed what I wanted to solve this problem, because J-Link is not a better way for dongles like Makerdiary and Feitian OpenSK dongle. 
> But this would break the key if it's configured with U2F support of batch attestation activated as the certificate and private key are also part of the storage.
Agree with this. 
> Updating the firmware will be done without DFU or with firmware signature activated
If you do not clear storage during flashing new firmware even though not over DFU, bad firmware can know User Credential. So verifying Signature is better to update the firmware.

jmichelp, 2020-12-22:
> If you do not clear storage when flashing new firmware no matter through DFU or J-LINK, bad firmware can know User Credential. So verifying Signature is better to update the firmware.

Sorry if this wasn't clear. I'll rephrase:
- either we decide to update firmware over DFU and then we must configure DFU to only accept signed images
- or we run our own firmware update mechanism as part of Tock/OpenSK. We need to design it but it will require signed images of course.

ia0, 2020-12-22:
I created https://github.com/google/OpenSK/pull/247 in case we want to go with the `erase_storage` example.

kaczmarczyck, 2021-11-19:
This is resolved with #247, and #414 adds the documentation to give `erase_storage` the necessary visibility.

closed 2021-11-19 by kaczmarczyck