google/OpenSK #194: Crypto materials should live in the persistent storage instead of the firmware

jmichelp, 2020-11-03:
The firmware currently contains:
- AAGUID
- a certificate for the key, signed by a dedicated CA
- a private key

We should be able to distribute a pre-compiled firmware which means that the certificate and the private key must be removed from the compilation time. It's fine to keep AAGUID in the firmware.

This means that we need to modify OpenSK to find these values as part of the persistent storage

ia0, 2020-11-03:
Those objects are accessed from the persistent storage since #112. However, we still need to embed the values in the binary until we can write them using vendor commands (tracked by #195) otherwise they are never set and the key is probably not usable. I think we can close this issue since it would be natural to remove the embedded values once they are set by the vendor commands.

jmichelp, 2020-11-03:
Ok, great.
Then marking as solved.

closed 2020-11-03 by jmichelp