google/OpenSK #195: Add vendor commands to inject crypto materials

jmichelp, 2020-11-03:
Once the crypto materials are removed from the firmware (i.e. when issue #194 is fixed), we need a way to inject these values into the FOB.
The easiest way to do so is by adding vendor specific commands and provide a command line tool to run them.

For security reasons, theses commands must only be executed once and the firmware should make sure to never overwrite a value that has already been programmed.

ia0, 2020-11-03:
Some code pointers regarding storage:
- Those [lines](https://github.com/google/OpenSK/blob/c9fc621f597e987035634aadac5c66f3e0b3fdcc/src/ctap/storage.rs#L234-L247) should be removed. Thus, the `key_material` module can be removed too.
- The vendor commands should use the `set_attestation_private_key`, `set_attestation_certificate`, and `set_aaguid` methods of the storage. If it is important that those are set at most once, I can write a PR for just that part since it's purely storage related.

jmichelp, 2020-11-03:
AAGUID will not be set through these commands. I'd like it to still be part of the firmware at the moment.
The certification is done for a specific AAGUID and a root CA so this makes sense to have AAGUID in the firmware. Then if you inject crypto materials that are signed by the corresponding CA you run a certified key otherwise not.

jmichelp, 2020-11-25:
List of pull requests required to achieve the goal:
- [ ] CTAP1/U2F accesses data through the store
- [ ] Vendor command is add to configure the authenticator (attestation crypto materials, JTAG lockdown)
- [ ] Capsule and syscalls are added to enable JTAG lockdown
- [ ] Configure tool to inject crypto materials in the authenticator
- [ ] Cleanup in the build script (e.g. removing openssl code/dependancy)

ia0, 2020-11-25:
Hopefully the first point is already fixed by #112. If it's not the case, let me know. I hopefully should be able to fix it.

kaczmarczyck, 2020-11-25:
We missed usage in [CTAP1](https://github.com/google/OpenSK/blob/master/src/ctap/ctap1.rs).

ia0, 2020-11-25:
I see @jmichelp just sent #222 for that :-) I don't know how I missed that usage.

PR #239 (2020-12-16): Add vendor commands to inject crypto materials
Fixes #195 

The PR is rather big and involves many modifications so I tried to keep the individual commits clean (they could be reviewed separately instead of the PR as a whole):
- Vendor command to configure the authenticator (attestation crypto materials, JTAG lockdown)
- Capsule, HIL and syscall to enable JTAG lockdown
- Configure tool to inject crypto materials in the authenticator
- Cleanup in the build script (e.g. removing openssl code/dependancy)
- Updated documentation
- Version bump from v0.1 to v1.0

Fully tested on the nRF52840DK boar [... cut]
closed 2020-12-16 by jmichelp