google/OpenSK #228: Chaning the certificate without rebuilding and re-flashing the firmware.

alishir, 2020-12-01:
Is it possible to change the certificate without rebuilding and re-flashing the firmware? I want to flash the firmware once and change the certificate periodically.

jmichelp, 2020-12-01:
No, at the moment it's not possible.
The certificate is only used if you enabled U2F/CTAP1 backward compatibility (enabled by default because not every website or platform is FIDO2/Webauthn ready) or if you enabled batch attestation in the source code (by default it's turned off). In all other cases, we use [`ATTESTATION_BASIC_SURROGATE`](https://fidoalliance.org/specs/fido-v2.0-rd-20180702/fido-registry-v2.0-rd-20180702.html#authenticator-attestation-types)

I'm currently working on removing the private key and the certificate from the firmware (see #195 ) so that the firmware can be compiled once (for a given `AAGUID`) and the certificate/private key can be loaded through vendor specific commands. But these operations are considered as part of a factory configuration that is a one-time operation. Any further attempt at loading a certificate of a private key will be rejected.

alishir, 2020-12-01:
Thanks a lot.

closed 2020-12-01 by alishir