google/OpenSK #240: How to install Google SK for below error

garywc0407, 2020-12-12:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu
info: Updating rust toolchain to nightly-2020-06-10
info: syncing channel updates for 'nightly-2020-06-10-x86_64-apple-darwin'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle_dfu
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.45.0-nightly (fe10f1a49 2020-06-02)`
    Finished release [optimized + debuginfo] target(s) in 0.02s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.04s
info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 939, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 734, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 634, in run
    self.build_opensk()
  File "./deploy.py", line 355, in build_opensk
    self._build_app_or_example(is_example=False)
  File "./deploy.py", line 391, in _build_app_or_example
    self.create_tab_file({props.arch: app_path})
  File "./deploy.py", line 407, in create_tab_file
    elf2tab_ver = self.checked_command_output(
  File "./deploy.py", line 295, in checked_command_output
    cmd_output = subprocess.run(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'elf2tab/bin/elf2tab'
root1@Garys-MBP opensk % ./deploy.py --board="nrf52840_dongle_dfu" --opensk --programmer=nordicdfu
info: Updating rust toolchain to nightly-2020-06-10
info: syncing channel updates for 'nightly-2020-06-10-x86_64-apple-darwin'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle_dfu
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.45.0-nightly (fe10f1a49 2020-06-02)`
    Finished release [optimized + debuginfo] target(s) in 0.02s
info: Building OpenSK application
    Finished release [optimized] target(s) in 0.03s
info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 939, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 734, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 634, in run
    self.build_opensk()
  File "./deploy.py", line 355, in build_opensk
    self._build_app_or_example(is_example=False)
  File "./deploy.py", line 391, in _build_app_or_example
    self.create_tab_file({props.arch: app_path})
  File "./deploy.py", line 407, in create_tab_file
    elf2tab_ver = self.checked_command_output(
  File "./deploy.py", line 295, in checked_command_output
    cmd_output = subprocess.run(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'elf2tab/bin/elf2tab'

## Specifications

- Version:
- Platform:

ia0, 2020-12-12:
Hi @garywc0407, did you try to run `./setup.sh` first? The output should look like [this](https://github.com/google/OpenSK/blob/master/docs/install.md#initial-setup).

Note that you cannot run `./setup.sh` twice without running `./reset.sh` in-between (which would revert all uncommitted changes).

garywc0407, 2020-12-12:
hi @ia0 , thanks i have reset once and setup again and it worked now!

ia0, 2020-12-12:
Cool, thanks for the feedback! I'm closing the issue then.

closed 2020-12-12 by ia0