google/OpenSK #241: Explicit use of big endian representation.

alishir, 2020-12-15:
Hi all,
According to FIDO2 specification all values are big endian encoded, the following quote taken from section 6 of FIDO2 specification:
> Messages from the host to authenticator are called "commands" and messages from authenticator to host are called "replies". All values are big endian encoded.

During reading HID module source code, I found the following line:
https://github.com/google/OpenSK/blob/02c9abc0a302a8e07789e0e37c688fb343f24fa1/src/ctap/hid/mod.rs#L238

Isn't it better to use `to_be_bytes` instead of `to_ne_bytes`?

kaczmarczyck, 2020-12-15:
We use `to_ne_bytes` at 2 places in the code: In `storage.rs` and to convert channel IDs. In both cases, the conversation just has to be consistent, it's not important for OpenSK how the number is interpreted. In this case, even though we allocate channels by incrementing `self.allocated_cids`, this is not a requirement by the specification, meaning the platform doesn't expect this behaviour.

It might make sense to only use `to_be_bytes` throughout the code to have unified behaviour and avoid future bugs. However, that would be a breaking change in storage (not so much for the channel ID). @ia0 thoughts?

ia0, 2020-12-15:
For the storage we have to use native endian because on nRF52 the smallest unit of write operations are words. The driver (Tock patch) uses native endian to convert from a word-aligned slice to the actual words written to flash. This is necessary to be able to read the slice from the storage as it was written without conversion.

Regarding the particular instance of `to_ne_bytes` in `src/ctap/storage.rs` it's just to have the kernel write the counter as a word of the actual value of the counter instead of swapped bytes. But this is not necessary, we could as well use `to_be_bytes` there. But that would be an unnecessary breaking change. I think it's better if the storage always uses native endian since it's what the processor uses to write to flash in the end.

kaczmarczyck, 2020-12-15:
@alishir I think the 4-byte sequence is the channel ID, not what integer the authenticator used to create the sequence. Do you have some issue in mind where it's important that platform and authenticator interpret the endianness? Otherwise, I don't have a strong opinion on the encoding used.

alishir, 2020-12-15:
No I didn't have any issue with that, only I think there is some semantic inconsistency between the code and the specification. For example if we have a channel with `CID = 2` we expect `0x00 0x00 0x00 0x02` in the raw packets, but I got `0x02 0x00 0x00 0x00` during raw packet inspections. I think using `to_be_bytes` for channel ID will helpful during debugging.

kaczmarczyck, 2020-12-16:
Got it. I created the PR making these changes, and will let @jmichelp decide if we want that or not.

kaczmarczyck, 2020-12-16:
(I'm leaning towards merging it.)

PR #242 (2020-12-16): changes channel ID endianness to big endian
Fixes #241 

To facilitate discussion, this is the PR that would fix #241, if we want to.

- [x] Tests pass
closed 2020-12-16 by kaczmarczyck