google/OpenSK #266: BLE support?

coco21, 2021-01-21:
Hello,
I was successful in making run the OpenSK full stack on a nRF52840 DK - great job guys!
I am wondering how easily can we make same stack run but in order to go over USB we go over BLE channel?
Did a lot of research work already and it seems Tock OS already BLE enabled somehow.

../OpenSK/third_party/tock/boards/nordic/nrf52840dk/src/main.rs, line #385

Namely:

    let ble_radio =
        nrf52_components::BLEComponent::new(board_kernel, &nrf52840::ble_radio::RADIO, mux_alarm)
            .finalize(());

I'd like to ask here effort this would be to make OpenSK work over BLE instead over USB as of today, what exactly must be done in a few words and if after all this is possible and makes sense. Already went checking at the Solo end as well who provide similar FIDO2 stack but in C and that seems quite easy to port on e.g. a native Nordic Zephyr OS - now I like your setup and it works.
Please advise quickly since I do this for an evaluation job where I've to take some decisions, thank you!

Best, Chris

jmichelp, 2021-01-21:
Hi Chris,
Tock only has support for BLE advertising. In order to use LBE for a security key, we would need an implementation of the Bluetooth Security Manager (the part of the protocol responsible for implementing device pairing, channel encryption, etc.). There is an ongoing attempt at improving this through the external library Rubble (https://github.com/tock/tock/pull/2233) but it hasn't landed yet.
At the moment we're pursuing our effort to support NFC in addition to USB transport protocol and don't have a plan to add BLE yet (and considering that NFC is now supported by iPhones and has a better user experience than BLE, I'm not sure why we would invest effort on BLE). But once NFC is supported in OpenSK and BLE with a Security Manager is supported by Tock, adding BLE shouldn't be too much difficult.

coco21, 2021-01-22:
Hey, Oh that was a quick answer, thank you very much, highly interesting information you give me here.
Fine I see. At my end I try to bring in maybe Nordic's (Thisisant) Softdevice S340 hex file into Tock, possible? Or better forget?
Thx Chris