google/OpenSK #268: lost access to bootloader on makerdiary nRF52840 MDK USB Dongle after flash

gled-rs, 2021-01-21:
## Expected Behavior
Holding button while plugging to computer enter UF2 flash mode ( ie presenting a mass storage device to the host )

## Actual Behavior
Holding button while plugging to computer does nothing

## Steps to Reproduce the Problem
1. generate  firmware with ./deploy.py --board=nrf52840_mdk_dfu --programmer=none --opensk
2. convert hex file to uf2 with uf2conv.py firmware.hex -c -f 0xADA52840
3. flash by copying the resulting uf2 file to the mass storage device presented.

## Specifications
- Version: b0c1b73897816edb757a163943f7b36a55f76a19
- Platform: linux

## Comments/Questions

Apologies if I missed something, I have a gut feeling I selected the wrong board, I chose nrf52840_mdk_dfu as the board since it was the only one mentioning mdk and I tried before nrf52840_dongle but the button was not recognized ( I was able though to enter UF2 mode with that firmware flashed ).

The dongle works perfectly as is, webauthn.io test is successful. I just can't set it back to bootloader mode to flash or update something else. Just want to confirm I need a debugger probe at that point to be able to update opensk and/or change the firmware running at that point. Thank you in advance for your pointers.

jmichelp, 2021-01-21:
Thanks for reporting.

This could be related to #186 

Probably unrelated but you could directly flash this dongle using DFU in one command:
`./deploy.py --board=nrf52840_mdk_dfu --programmer=nordicdfu --opensk`

Once the bootloader becomes unaccessible, you indeed need to use a JTAG probe on the PCB. I don't know any other method to restore it at the moment.

I will try to find a Makerdiary dongle without the case so that it's easier for me to run some tests on it without having to first destroy the plastic case :)

gled-rs, 2021-01-21:
> Thanks for reporting.
>
> This could be related to #186
>
> Probably unrelated but you could directly flash this dongle using DFU in one command:
> `./deploy.py --board=nrf52840_mdk_dfu [...]

Thank you, I placed an order for a jlink to go further and be able to reflash at will, I will be able to provide more testing at that point.

FWIW, the case split very nicely with a tiny flat head, it is not glued together, just pins snapping in place ( I tried to short reset / gnd just in case, did not work ).

jmichelp, 2021-01-21:
I had a quick look at the bootloader source code (I assumed the Makerdairy dongle was using the Adafruit UF2 bootloader).

Bootloader parameters are stored in the flash, at a place that is very likely to be used either by Tock OS kernel or OpenSK code:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/linker/nrf52840.ld#L19-L22

So it's very likely that the parameters got overwritten and that it's the root cause for not being able to enter the bootloader anymore.

gled-rs, 2021-01-21:
I have a second MD dongle if you want me to do some tests, or should receive the jtag debugger in about a week.

asmw, 2021-02-08:
I was hit by this and reflashed the bootloader via JTAG (RIP plastic case).
If required I can also test.

jmichelp, 2021-02-11:
And after reflashing the bootloader, did both OpenSK and the bootloader worked?

asmw, 2021-02-11:
@jmichelp no, I immediately see the U2F drive now when I plug the dongle in.
But I switched bootloaders from OpenBootloader to the U2F bootloader, so that might play into that.

asmw, 2021-02-12:
Both OpenSK and the uf2 bootloader work when I use [the makerdiary OpenSK master, rebased onto the current develop](https://github.com/makerdiary/OpenSK/tree/master)
I.e. if I plug in the dongle, I see the OpenSK HID device in dmesg, if I plug it in with the reset button pressed, the UF2 drive appears.

As far as I can tell the only functional change in the makerdiary master branch is the one from PR #186

asmw, 2021-02-12:
Just tested: I cannot register on webauthn.io with the working combo (a 'real' fido key works)

When I press register the dongle flashes in multiple colors (I guess that's the prompt?) Pressing the button interrupts the blinking briefly, but nothing else happens. The browser prompt stays.

How can I debug this?

axmmisaka, 2021-02-23:
asmw: did you build with `--board=nrf52840_mdk_dfu`? (assuming you are using that)
once you use nordic tool to erase the persistent memory (make sure you do that as well) & flash in openbootloader/UF2BootLoader behaviour should be the same as a brand-new one

cxgreat2014, 2021-02-26:
> Just tested: I cannot register on webauthn.io with the working combo (a 'real' fido key works)
>
> When I press register the dongle flashes in multiple colors (I guess that's the prompt?) Pressing the [...]

Similar question for me, it works good at the beginning, I can use it to register and login website as yubikey, but after some login and register action,  in the login action the dongle flashes in multiple colors , I push the button it interrupts the blinking, but the browser prompt stays.

I flash it again but did not work, and I lost all my key.... 😢

asmw, 2021-02-26:
@axmmisaka My build command is `./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none`

How can I erase the persistent memory?

axmmisaka, 2021-02-26:
There should be a erase option in nRF Connect after you connect your board via JTAG. That said though, I am not entirely sure if that would fix your problem (I erased it and it did not cause problem)

jmichelp, 2021-08-16:
Closing stale issue, assuming the problem was solved.

closed 2021-08-16 by jmichelp